davesteele/comitup #99: Not List SSID

veyselolgun, 2020-02-06:
Hello,
I am using Comitup 1.3.1-1 and 
I am able to connect any network (via sudo comitup-cli ) (in Raspberry)
I am trying to monitoring all ssid in network but it is not monitoring 
Is this relevant comitup Does anyone have an idea ? 

`elseif($islem == 'connect') {
                                    ?>
                                    <div class="container">
                                        <div class="row">
                                            <div class="mtop40">
                                                <div class="col-md-4 col-md-offset-4 text-center">
                                                    <h1 class="text-uppercase mbot20">WiFi Connection</h1>
                                                </div>
                                            </div>
                                        </div>
                                        <div class="container">
										<? if ($_GET['ssid'] <> '') { ?>
                                            <div class="col-md-4">
                                                <h3>Connection to <?=$_GET['ssid'];?></h3>
                                                <p>
                                                    Type the password and press the Connect button to connect this device to <b><?=$_GET['ssid'];?></b>.<br>

                                                    This will terminate the current hotspot and try connecting to <b><?=$_GET['ssid'];?></b> for access.<br>

                                                    If the connection fails, this hotspot will be re-created.<br>
                                                </p>
                                                <form method="post" action="?islem=connect">
                                                    <input type="hidden" name="ssid" value="<?=$_GET['ssid'];?>">
                                                    <div class="control-row">
                                                        <label>Password : </label>
                                                        <input type="password" name="password" class="form-control">
                                                    </div>
                                                    <div class="control-row" style="margin-top: 10px; padding-top: 10px;">
                                                    <input type="submit" value="Connect" class="btn btn-info">
                                                    </div>
                                                </form>
                                            </div>
                                            <? } ?>
                                            <? if ($_POST['ssid']){ ?>
                                                <h3>Connecting to : <?=$_POST['ssid'];?></h3>
                                                <hr>
                                                <?
                                                $url =  'http://'.$_SERVER['HTTP_HOST'].':81/connect';
                                                $ch = curl_init();
                                                curl_setopt($ch, CURLOPT_VERBOSE, true);
                                                curl_setopt($ch, CURLOPT_URL, 'http://10.42.0.1:81/connect');
                                                curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);
                                                curl_setopt($ch, CURLOPT_POST, true);
                                                curl_setopt($ch, CURLOPT_POSTFIELDS,
                                                    "ssid=".$_POST['ssid']."&password=".$_POST['password']);
                                                $output = curl_exec($ch);
                                                curl_close($ch);
echo $output;
                                                ?>
                                            <? } else { ?>
                                            <div <? if ($_GET['ssid'] <> '') { ?>class="col-md-6"<? } ?>>
                                               <div class="row"><div class="col-md-6"><h4>SSID</h4></div><div class="col-md-6"><h4>Action</h4></div></div>
                                                <?php
                                            $url =  'http://'.$_SERVER['HTTP_HOST'].':81/';
                                            $ch = curl_init();
                                            curl_setopt($ch, CURLOPT_VERBOSE, true);
                                            curl_setopt($ch, CURLOPT_URL, 'http://10.42.0.1:81/');
                                            curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);
                                            $output = curl_exec($ch);

                                            curl_close($ch);

                                            $lines = explode('#####',$output);
                                            for ($i=0; $i<count($lines)-1;$i++){
                                                echo '<div class="row" style="margin-top: 5px;"><div class="col-md-6">'.$lines[$i].'</div><div class="col-md-6"><a class="btn btn-success" href="?islem=connect&ssid='.$lines[$i].'">CONNECT</a></div></div>';

                                            }
                                            ?>
                                            </div>
                                            
                                            <? } ?>
                                    </div>
                                    <?
                                }`

davesteele, 2020-02-09:
NetworkManager has trouble maintaining the available AP list via an interface which is serving an AP. The list is more complete if you add another Wifi interface using a USB device.

veyselolgun, 2020-04-13:
Thank you, i will keep this. And after COVID-19 i will try this (Because no work currently)

closed 2020-06-17 by davesteele