davesteele/comitup #101: Wifi wlan concern

mrsstaceyseigle, 2020-02-16:
I am planning on using a raspberry pi compute module 3 and a custom IO pcb board. The IO does not have built it wifi or Ethernet support. I play to use a usb wifi adapter and its the only wlan network interface. Should I have any problems with your software seeing and using the usb wifi adapter as default wlan device. I also play to use a usb Ethernet as well

davesteele, 2020-02-17:
It will work, if the USB Wifi adapter and associated driver support Access Point mode. Most do not.

My latest go-to adapter is [this guy](https://www.amazon.com/gp/product/B016I6DJ5C).

I'd make sure that there is an entry in /etc/network/interfaces for the Ethernet device, with an appropriate configuration.

mrsstaceyseigle, 2020-02-17:
Great. The adapter that you are referring do it support the access point mode needed for your software

closed 2020-03-11 by mrsstaceyseigle