davesteele/comitup #102: Debian buster repo version doesn't work on Raspberry Pi

schweini, 2020-02-16:
On Buster on a Rapi3, the official repo version fails to bring up a hotspot (windows just show a ;hidden network').
The updated version from your repo works fine, though.

I'd recommend updating the debian repo version, because it is quite a PITA that it leaves the RaPi in a disconnected state, which is a nuisance if it is headless (like mine was).

davesteele, 2020-02-17:
Buster is Debian's current stable release. It's non-trivial to get changes into that version in a way that is generally visible to users.

The [Comitup Image](https://steele.debian.net/comitup/) is a Raspbian Buster Spin with a current version of comitup installed. It is the recommended path to working with Comitup. The [home page](https://davesteele.github.io/comitup/) also points to how to get the latest package installed.

I've [updated](https://github.com/davesteele/comitup/commit/42193a23be006549a9566660858fc6d224887e24) the home page to make this clearer.

schweini, 2020-02-17:
Thank you for the changes! Although - if I am not the only one who gets a inaccessible system upon installing comitup on Buster - I personally would phrase it a bit stronger than a "recommendation". 

(but then again, I'm just bitter because I didn't have a fitting monitor at hand to connect to the RasPi)

Thanks for the very useful software, anyhow!

closed 2020-03-17 by davesteele