davesteele/comitup #103: Comitup causing very slow boot

Jesse-CProof, 2020-03-04:
Raspberry Pi Zero W, Raspbian Buster, Comitup v1.10

Installed Comitup from the repo, cleared my wpa_supplicant.conf and disabled systemd-resolved.service.

systemd-analyze blame shows anywhere from 20-40 seconds (24.231 for the below case) for comitup.service to run at boot. Is this normal?

I currently have `nohook wpa_supplicant` in my dhcpcd.conf file in an attempt to help the situation, but it didn't seem to have much of an effect.

The comitup log for my last boot looks like this, showing it's setting up a hotspot before connecting to the saved network. I thought it tried to connect before establishing the AP?:
```
2020-03-04 22:38:25,204 - comitup - INFO - Starting comitup
2020-03-04 22:38:29,272 - comitup - INFO - Setting state to HOTSPOT
2020-03-04 22:38:29,291 - comitup - INFO - Activating hotspot
2020-03-04 22:38:42,698 - comitup - INFO - Setting state to CONNECTING
2020-03-04 22:38:47,051 - comitup - INFO - Attempting connection to BrewNet
2020-03-04 22:38:53,268 - comitup - INFO - Setting state to CONNECTED
```

Anything I can do to speed this up? For this device, time-to-boot is a pretty important metric. Thanks in advance!

davesteele, 2020-03-05:
Comitup is defined to be a [dbus service](https://github.com/davesteele/comitup/blob/b00ce26fd11862a0b6cf065e3956b84f270779d8/debian/comitup.service#L8) - it's not deemed to be up until the dbus interface is available. It doesn't do that until network issues are set up, and the underlying NetworkManager is known to be sluggish at startup.

You could change the type to "simple", and there would be no reported delay from systemd. comitup-web would probably have a problem with this, but you might be able to avoid the problem by removing the [two immediate dbus calls](https://github.com/davesteele/comitup/blob/b00ce26fd11862a0b6cf065e3956b84f270779d8/web/comitupweb.py#L135) in the web startup, and hope that no-one gets on the web page too quickly.  IIRC, I added those back when I had an initialization problem with the comitup server.

I just did a quick test of this, and it appears to work fine.

You probably need to run "systemctl daemon-reload" after the service file change.

davesteele, 2020-03-05:
Looking more closely at your question, this probably won't help you. It will still take time to make it to the CONNECTED state.

Jesse-CProof, 2020-03-05:
Thanks for the quick response. It sounds like the program is behaving as intended, so we can go ahead and close this one out. I'll have to see if there are other ways I can speed things up.

closed 2020-03-05 by Jesse-CProof