davesteele/comitup #105: USB device noise when comitup run

hakao32, 2020-03-16:
I have a USB soundcard for my RPi 3.
http://60.248.10.74/support/download/542/UDA-500(Spec).pdf
It works great, I can get the Mic input clearly but when I start comitup service a noise rise. And noise doesnt disapper even if I stop comitup service again.
What kind of services comitup use? I think it is about WiFi search because when I put my soundcard away from RPi with usb expander, noise disappears.

hakao32, 2020-03-16:
And while service running bot connected to a WiFi, there is no noise. Noise only appears in HOTSPOT mode.

davesteele, 2020-03-16:
It is likely the wifi. To verify:

- Stop comitup (sudo systemctl stop comitup)
- Manage the AP using the NetworkManager nmcli command
    sudo nmcli connection
    sudo nmcli connection {up|down} &lt;connection name&gt;

Does this turn the sound on/off?

If so, why does the AP cause noise and not a normal connection? Possibilities - the AP is limited to 2.4GHz, while your other connection may be 5GHz. The limitation in AP is caused by the "band=bg" line in the NM configuration file (in /etc/NetworkManager/system-connections). APs are always sending traffic - does the noise come up with an upstream connection if the link is busy? It may also be the channel - change it with "channel=&lt;n&gt;" in the wifi section of the config file.

hakao32, 2020-03-17:
sudo systemctl stop comitup
sudo nmcli connection down <connection name>
I did this there is no noise at all when AP down. But if I up AP, the noise comes back.
How I can understand if link is busy or not. Noise come up only at AP no matter if any other device connected to the AP or not. There is no 5GHz device nearby.
Which config file should I put "channel=<n>" ?

davesteele, 2020-03-17:
busy = use it to watch YouTube
I was suggesting that your upstream connection may be using 5GHz.
Channel= is part of the NM config, in the wifi section. See  https://developer.gnome.org/NetworkManager/stable/settings-802-11-wireless.html.

You have an interference problem between your USB devices. You need to isolate or replace them.

closed 2020-06-09 by davesteele