davesteele/comitup #108: Install in Raspbian and Retrieve IP

CarloPSW, 2020-05-03:
Hi Steve, great job.  I tried the comitup image on a rasp and it works perfectly!
  I have to install comitup on existing raspbian, I try to follow the installation in the guide but I can't see the Wi-Fi comitup.  The network management service is not running.  How can I complete my installation?

  I have another request, is it possible to see which IP has been corrected after the correct connection to an existing Wi-Fi?

Thank you!

davesteele, 2020-05-03:
As far as I know the package [installation instructions](https://github.com/davesteele/comitup/wiki/Installing-Comitup) on the wiki are still good.

The easiest way to get the IP address is via a [Zeroconf Browser](https://github.com/davesteele/comitup/wiki/ZeroconfBrowser). We can't count on the old hotspot being available to show the address that way.

CarloPSW, 2020-05-03:
> As far as I know the package installation <https://github.com/davesteele/comitup/wiki/Installing-Comitup> on the wiki are still good.

Hi, I followed all the steps, restarted raspberry, but nothing. I think the problem is the service, I have not understood if it is active and must be stopped or I have to start it.

I renamed the wpa file, but nothing. Where am I doing wrong?

as for the second point, it would be nice. tasmota's wifimanager can figure out which IP has been assigned.

Thank you so much.

CarloPSW, 2020-05-03:
Ok, all done! Thank you! 

Just last questione, how can i start automaticaly the web browser when in Connected ti comitup Wi-Fi?

davesteele, 2020-05-03:
I'm assuming you mean web server. See [man 5 comitup-conf](https://davesteele.github.io/comitup/man/comitup-conf.5.html)

CarloPSW, 2020-05-04:
Perfect! Thank you!

closed 2020-06-09 by davesteele