davesteele/comitup #109: It is not a issue

santhoshnp, 2020-05-04:
I have one doubt everything working perfectly for me. but still i have one doubt. WiFi display as comitup-n OK i have choose the WiFi and it open in the browser as 10.41.0.1 but it has no ssh certificate. how to give ssh certificate for website ( where we can access our desire wifi)

davesteele, 2020-05-04:
Can't be done in the way you suggest. Tls certificates are signed by a trusted third party, and that won't happen for a private address like 10.41.0.1. There is a "self signed"option, but that can be hijacked, providing a false sense of security.

santhoshnp, 2020-05-04:
Thank you so much for the explanation. Really your community is doing excellent job all the best for the future.

closed 2020-06-09 by davesteele