davesteele/comitup #113: How to debug missing DHCP IP?

marcelstoer, 2020-06-03:
While testing tonight I noticed on a few occasions that client devices didn't get a DHCP IP from the Pi when it was in AP mode. The client (MacBook & iPhone) joined the WiFi AP `comitup-nn` just fine but didn't get an IP. 

Once I manually set the IP to 10.41.0.2 plus subnet mask plus router IP 10.41.0.1 the comitup-web interface was accessible. 

I checked the FAQ, the troubleshooting guide and the issues list but didn't find any hints. What could be the root cause and how could I debug this?

davesteele, 2020-06-04:
DHCP is provided by dnsmasq, which is configured by the conf/dns-*.conf files, and controlled by cdns.py. The most likely cause is that there is a competing DHCP service on the address/DHCP port for the AP. 

sudo netstat -elnp | grep 67

If true, there will be errors visible in the systemd log for comitup.

Cwmwd24, 2020-06-04:
I was having similar problems with comitup in router mode not handing out IP addresses on the hotspot so could not connect.

Nothing obvious in the system logs but `systemctl status comitup` had errors like these:

```
Jun 04 22:48:40 raspberrypi comitup[1503]: dnsmasq: failed to create listening socket for port 53: Address already in use
Jun 04 22:48:40 raspberrypi dnsmasq[1586]: failed to create listening socket for port 53: Address already in use
Jun 04 22:48:40 raspberrypi dnsmasq[1586]: FAILED to start up
Jun 04 22:48:40 raspberrypi comitup[1503]: dnsmasq: failed to create listening socket for port 53: Address already in use
Jun 04 22:48:40 raspberrypi comitup[1503]: dnsmasq: failed to create listening socket for port 53: Address already in use
```
Eventually fixed this by saying `sudo systemctl disable dnsmasq` and restarting comitup. I'd also tried `systemctl disable dhcpcd` in the process of trying to resolve the problem but I'm not sure if it's relevant.

I'm using "image_2020-04-05-Comitup-lite" on a Raspberry Pi 3 Model A+ with a USB WiFi dongle for the station.

marcelstoer, 2020-06-05:
I installed the comitup package to a recent (~3w old) Raspian on a Pi 4. Here's the output I get after restart when the device is in client mode:

```
pi@pink-panter:~ $ systemctl status comitup
● comitup.service - Comitup Wifi Management
   Loaded: loaded (/lib/systemd/system/comitup.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-06-04 17:57:40 CEST; 13h ago
     Docs: man:comitup(8)
 Main PID: 498 (comitup)
    Tasks: 1 (limit: 4035)
   Memory: 25.1M
   CGroup: /system.slice/comitup.service
           └─498 /usr/bin/python3 /usr/sbin/comitup

Jun 04 17:57:41 pink-panter dnsmasq[584]: FAILED to start up
Jun 04 17:57:41 pink-panter dnsmasq[589]: started, version 2.80 cachesize 150
Jun 04 17:57:41 pink-panter dnsmasq[589]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect 
Jun 04 17:57:41 pink-panter dnsmasq-dhcp[589]: DHCP, IP range 10.41.0.100 -- 10.41.0.150, lease time 10m
Jun 04 17:57:41 pink-panter dnsmasq[589]: reading /etc/resolv.conf
Jun 04 17:57:41 pink-panter dnsmasq[589]: ignoring nameserver 127.0.0.1 - local interface
Jun 04 17:57:41 pink-panter dnsmasq[589]: read /etc/hosts - 5 addresses
Jun 04 17:57:47 pink-panter dnsmasq[589]: exiting on receipt of SIGTERM
Jun 04 17:57:47 pink-panter comitup[498]: OK
Jun 04 17:57:47 pink-panter comitup[498]: OK

pi@pink-panter:~ $ sudo netstat -elnp | grep 67
udp        0      0 0.0.0.0:68              0.0.0.0:*                           0          14276      674/dhclient        
raw6       0      0 :::58                   :::*                    7           0          13967      861/dhcpcd          
unix  2      [ ACC ]     STREAM     LISTENING     13167    861/dhcpcd           /var/run/dhcpcd.unpriv.sock
```

muelli, 2020-06-05:
just guessing: Is systemd-resolved running? Cf. https://github.com/davesteele/comitup/issues/51

marcelstoer, 2020-06-05:
Good point; I added the `sudo netstat -tulpen` output to the above comment.

davesteele, 2020-06-05:
I've made new images, with the dnsmasq service masked (one step past disabled). Does this help?

marcelstoer, 2020-06-05:
```
pi@pink-panter:~ $  sudo apt-get install comitup
Reading package lists... Done
Building dependency tree       
Reading state information... Done
comitup is already the newest version (1.10-1).
```
Which is in line with what I see here https://davesteele.github.io/comitup/archive.html and here https://github.com/davesteele/comitup/tree/gh-pages/deb

davesteele, 2020-06-05:
New disk images.

marcelstoer, 2020-06-05:
Sorry, got it. Still getting familiar with the project and the Pi world.

I masked it manually and rebooted the Pi. Then I cleared the comitup/NetworkManager connection. Connecting to the WiFi and getting an IP worked reliably. I tested this a couple of times.

davesteele, 2020-06-05:
I've added [detail](https://github.com/davesteele/comitup/wiki/Installing-Comitup/_compare/0f8ca082c1336461d4d3b4e5a2be8ceb8a78b390...c36ed739022b0e3a3b53cec360408aa2a2c0b243) to the Wiki install page, and also modified the disk image to address this issue. Closing pending additional input.

closed 2020-06-05 by davesteele