davesteele/comitup #114: Q: under what circumstances does the hotspot start & terminate?

marcelstoer, 2020-06-05:
Sorry if this is obvious but I didn't see it mentioned anywhere.

Scenario 1: a _known/saved_ WiFi is temporarily unavailable upon startup, the comitup hotspot starts. 
- Do the WiFi connection attempts continue in the background even while the hotspot is active?
- Does the hotspot terminate automatically as soon as (one of) the saved WiFis becomes available again?

Scenario 2: the Pi looses connection to the WiFi it connected to upon start.
- Does the comitup hotspot start even now or only if the Pi can't connect to a WiFi at startup?

davesteele, 2020-06-06:
Scenario 1: a known/saved WiFi is temporarily unavailable upon startup, the comitup hotspot starts.

    Do the WiFi connection attempts continue in the background even while the hotspot is active?
        Yes. It is a periodic check.
    Does the hotspot terminate automatically as soon as (one of) the saved WiFis becomes available again?
        If you have a single WiFi adapter, then yes. That adapter is used for the new connection.
        If you have two WiFis, then both connections can remain. Appropriate routing is automatically
           set up between them. See "man 5 comitup.conf".

Scenario 2: the Pi looses connection to the WiFi it connected to upon start.

    Does the comitup hotspot start even now or only if the Pi can't connect to a WiFi at startup?
        Yes, it starts in this situation as well.

marcelstoer, 2020-06-06:
All clear, thanks. That's exactly how I expect the system to behave; POLA.

closed 2020-06-06 by marcelstoer

alf4all, 2022-02-20:
On a Raspberry Pi 3B with OS 32-bit I have installed apache2 before installing comitup. Now I can access the apache2 default page via WiFi but how can I make a setup to an existing AP? I it by a webpage or by SSH? did I miss something?