davesteele/comitup #119: Activate ap_password

abdessamade14, 2020-07-07:
Hello, 
I have read the comitup-conf(5) because i would like to add a password in order to acces to the hotspot.
So I add in the comitup.conf     ap_password: "password" just after ap_name.
There is no problem for the name it change but when i reboot and i try to connect to the hotspot.
It didn't ask the password that i add before.

I try with "password" and password but there is no difference.

Could you help me please

davesteele, 2020-07-08:
The configuration file for the Network Manager connection created for the hotspot is stored in /etc/NetworkManager/system-connections. Take a look that that, [this](https://developer.gnome.org/NetworkManager/stable/nm-settings.html), and the nmcli command ("nmcli con help") for working with NM connections.

abdessamade14, 2020-07-08:
> The configuration file for the Network Manager connection created for the hotspot is stored in /etc/NetworkManager/system-connections. Take a look that that, [...]

Thank you for your answer. I have checked the nmcli command as you said and i finally update my password with some comand line.
Thank you

closed 2020-07-08 by abdessamade14