davesteele/comitup #121: wifi connection problem

asadasivam, 2020-08-03:
Hi team,

I have two routers. So First I have connected in one router by using comitup os(raspberry pi)it is connected successfully and I power off the router and  tried to connect with another router it is not connecting it shows authentication error I don't why can you please
explain to me about this I am so enthusiastic to learn about this

davesteele, 2020-08-03:
I don't think the fact that you have two routers involved has anything to do with the problem (unless they have the same ssid).

Does NetworkManager have anything to say about the failure? (sudo grep NetworkManager /var/log/user.log).

Note that the configuration for the connection is stored in /etc/NetworkManager/system-connections. You can work on details from there.

Also note that you can invoke the connections from NetworkManager itself:

sudo systemctl stop comitup
sudo nmcli con up &lt;ssid&gt;

asadasivam, 2020-08-04:
Thank you so much for the response, but it showing

asadasivam, 2020-08-04:
It is showing like this 
[image]

asadasivam, 2020-08-04:
can you please explain about this

davesteele, 2020-08-04:
Not really. The connection shows as open, meaning no password is needed. Maybe you're using a USB Wifi for the hotspot which is not well supported by Raspbian.

Comitup implements a "captive portal", using dnsmasq. Perhaps logs for that utility may give some insight.

asadasivam, 2020-08-04:
No, I am not using USB wifi for the hotspot, so what will be the reason

asadasivam, 2020-08-04:
it says that couldn't connect to the network why?

davesteele, 2021-02-13:
i encountered this problem. Resolved with https://github.com/davesteele/comitup/commit/9c044dcd9d436876cdfc0f9ce0948520281901cc, released with v1.15, and the latest image.

closed 2021-03-15 by davesteele