davesteele/comitup #122: Need Diet Pi image also

asahajit, 2020-09-12:
As Diet pi us is the best choice for headless setup.
I found that Comitup conflicts with the standard Debian WiFi configuration, and could not set up a configuration such that Comitup would run on DietPi. It proved more efficient to go with the regular Comitup image, despite the full Raspbian that came with it. Regardless, these are both great projects; if anyone manages to combine them or finds another provisioning portal for DietPi, please let me know!

davesteele, 2020-09-12:
You should know that Comitup pulls in a fair number of packages, including NetworkManager, and still easily fits on the smallest SD card you'll find in the store. If Raspbian Lite feels bloated to you, you may want to find another way.

As far as your specific problem goes, the top three reasons for installation problems is 1) a conflict in who is providing DHCP service, 2) a conflict in who is providing wpa_supplicant service, and 3) NetworkManager is not being permitted to manage the Wi-Fi adapters. This is addressed in the [Wiki](https://github.com/davesteele/comitup/wiki/Installing-Comitup)

muelli, 2020-09-13:
I bet this is https://github.com/davesteele/comitup/issues/51.  It could be fairly simply fixed, but hasn't yet. @asahajit you may have luck looking in that bug for ways to make it work.

davesteele, 2020-09-14:
"[I from my side will not touch such high level tools but concentrate on writing our own lightweight DietPi-style implementations ](https://github.com/MichaIng/DietPi/issues/3773#issuecomment-691505773)"

I agree. Closing

closed 2020-09-14 by davesteele