davesteele/comitup #124: Unable to connect browser to IP address 10.41.0.1

bobrathbone, 2020-11-19:
Platform: Raspberry Pi 3B:  Raspberry OS Version 10 (Buster)  Kernel version 5.4.72, comitup version 1.13

I have previously installed and used comitup on Raspberry Pi without problems and have been very pleased with the facility it offers. However, on the last two occasions I am unable to connect to http://10.41.0.1 despite the fact that  the installation of comitup went without problem. 

I see IP address 10.41.0.1 on interface  WLAN0 and can ping it from the local Raspberry Pi. 
I can also telnet to 10.41.0.1 port 80 and issue a get command, so I know that comitup-web is running. on 0.0.0.0 port 80
I have confirmed that comitup(-web) is running on port 80 and 53. If I stop comitup(-web) the entries shown below by netstat disappear.
```
$ netstat -tan
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN
```
**wpa_supplicant.conf** has been moved out of the way and **dnsmasq.service** has been both disabled and masked.
I have disable **apache2.service** and have added the following line to the **/etc/comitup.conf** file 
`web_service: apache2.service`
Despite all this I am not able to connect to comitup-web on 10.41.0.1 nor ping this hotspot IP address from any device on my network.

I ran comitup-cli and it shows that the hotspot is enable.
```
$ comitup-cli
State: HOTSPOT
Connection: comitup-132
```
I was able to connect to my router (EE-Hub-5G5z) using comitup-cli without problem
```
State: CONNECTED
Connection: EE-Hub-5G5z
```
I have reached the point where I cannot think of what other things I can try nor can I see what I might have missed.
I would be grateful if you can provide any guidance. I attach various dumps I have made for additional information.

Regards
Bob Rathbone
[www.bobrathbone.com](url)

[Comitup notes - Bob Rathbone.txt](https://github.com/davesteele/comitup/files/5566920/Comitup.notes.-.Bob.Rathbone.txt)

davesteele, 2020-11-21:
I'm not confident the referenced commit fixes your problem, but it might.

Beyond that, I am not seeing the problem. Note that I am using the Comitup Raspberry Pi OS image. You can inspect the [image generation process script](https://github.com/davesteele/pi-gen/tree/comitup/stage2/04-comitup) to get an idea of what may be different from your environment.

I'm also not running an apache web service. Does disabling that help?

bobrathbone, 2020-11-28:
I tested with the above commit but no difference. I then downloaded and installed the Comitup Raspberry Pi OS (Buster) image but got the same result. I attempted to both ping and telnet (port 80) to 10.41.0.1 from another Raspberry Pi configured on same WiFi network (wlan0) on a 192.168.0.0/24 network.  Using **tcpdump** I checked for incoming requests on interface wlan0 with the address 10.41.0.1 on the Comitup Raspberry Pi but did not see any incoming packets. I can only assume that there is something peculiar about my environment although I am hard pressed to think what. I will try testing at another location to confirm this.

davesteele, 2020-11-28:
I just tried replicating your problem with a pair of Pi's, and cannot replicate the problem.

davesteele, 2021-04-23:
Just noticed the report is against 1.13. See [CONTRIBUTING](https://github.com/davesteele/comitup/blob/main/CONTRIBUTING.md).

closed 2021-04-23 by davesteele