davesteele/comitup #125: DHCP server on Ethernet ?

mstormi, 2020-11-20:
Apparently comitup serves dhcp on the interfaces it works on. Makes sense for Wifi but for Ethernet ?
I had a comitup box connected via LAN, too, and it messed up my home network overruling my regular DHCP server
Had a power outage with everything booting again and some of my clients connected to the comitup box which is
definitely not desired (in probably not use case whatsoever) so can you disable serving DHCP to Ethernet please ?

davesteele, 2020-11-21:
Try those edits. The files are in /usr/share/comitup/dns/.

mstormi, 2020-11-21:
Thanks for your fast response.
However, turns out that whole "dns" directory does not exist on my box.
I did not use the image but manually installed the package - would it need to nonetheless exist ?
I'm using the comitup package from and on top of latest Raspi OS, v1.3.1-1 according to dpkg -s but frankly I don't know for sure what comitup version is contained and comitup seems not to have a -version switch (so how can I find out ?)

davesteele, 2020-11-21:
For the version:

"dpkg --list comitup"

For the location of the config files in your setup:

"dpkg --listfiles comitup"

There are a couple of ways to stay with more recent versions. Use the [Comitup spin](https://github.com/davesteele/comitup/wiki/Tutorial#download-the-comitup-image), or add the [Comitup Repository](https://davesteele.github.io/comitup/ppa.html).

davesteele, 2020-11-21:
The version in Buster, 1.3.1-1, is 2 years old. There have been many, many releases since then (1.13-4 is current).

mstormi, 2020-11-22:
oh ok. Got 1.13 working fine now. Apologies I didn't check before raising this issue. Thanks for your support.

closed 2020-11-22 by mstormi