davesteele/comitup #126: Setting ap_password in a pi-gen image

depthoffocus, 2020-11-23:
Hi,

Comitup is fantastic, thanks so much for this!

I'm setting up a headless device to help a set of potentially non-technical users. 

I'm building my image using pi-gen, and have set up a stage2 step patterned on yours, which I understand fairly well -- I guess you do all the post-install work in 02-run.sh because when running in qemu, comitup doesn't see the network devices to configure?

All works really well -- except I'm struggling with ap_password. I've had a look at #119 but it's not really shedding too much light on what I will be able to do.

After your 02-run.steps, I copy in my comitup.conf, with ap_password set, but the machines booted on this image don't insist on the password.

Is there something I am missing that I can do within pi-gen? (there probably is :-) )

Thanks again for amazing work!

davesteele, 2020-11-23:
02-run.sh currently just sets up eth0 to automatically create a DHCP or non-DHCP connection, as appropriate. It needs to have NetworkManager installed first.

You want to edit or overwrite [/etc/comitup.conf](https://davesteele.github.io/comitup/man/comitup-conf.pdf) after comitup is installed, which you say you are doing. 

Are you sure it is being copied to the right place?

Do the NetworkManager settings files (/etc/NetworkManager/system-connections/comitup*) created for the hotspot include "802-11-wireless-security" entries for key-mgmt and psk?

You should expect the NM settings files to be overwritten at comitup's whim.

depthoffocus, 2020-11-23:
Yes. I'm basically using your stage2 script with some modifications, and after your lines in 02-run.sh I have my own:

```bash
#!/bin/bash -e

for conn in dhcp static; do
    install -m 600 files/${conn}.nmconnection ${ROOTFS_DIR}/etc/NetworkManager/system-connections/
done

install -m 600 files/comitup.conf ${ROOTFS_DIR}/etc/
```

I have this sort of line in there:

```
ap_password: passw0rd! 
```

This config file definitely gets picked up in the Pi at boot, because it also renames the network, and that works fine.

In answer to your second question, I figured that was the issue you were pointing at in #119, but no -- no `802-11-wireless-security` settings, but (from memory) the `802-11-wireless` settings are all there.

depthoffocus, 2020-11-23:
Mm actually, is it as simple as having chosen a bad password, I wonder -- perhaps NM refused to use it.

depthoffocus, 2020-11-23:
Yes... It's probably that. The one I actually chose was too short. I will test that out.

depthoffocus, 2020-11-23:
Yes! 

Sorry about that. That was the cause. :)

closed 2020-11-23 by depthoffocus