davesteele/comitup #127: Issue with pi hole

Veeramk, 2020-11-23:
Hello, I have comitup  running on my raspberry pi smoothly. But as I add pi hole to it when I try to connect it wide it showed the pi hole page. I am bit of a newbie so please help me with this. By the way awesome work. Thank your

davesteele, 2020-11-23:
Comitup and pi-hole are competing for resources. At a minimum that includes DNS and web services. Take a look at the [comitup-conf](https://davesteele.github.io/comitup/man/comitup-conf.pdf) man page and the web-service keyword. If you can classify the pi-hole activities as a systemd service, you could use that flag to get the two to co-exist.

I wouldn't classify this as a newbie task. If you get it to work, I'd love to see a write-up.

Anjum9694, 2021-02-26:
First you would need to change the port number Pi-hole lighttpd uses in `/etc/lighttpd.conf` to something other than 80. This will be the port you will use to access the admin page.

Then add a delay to the startup of the Pi-hole DNS server by adding `DELAY_STARTUP=120` to `/etc/pihole/pihole-FTL.conf`
2 minutes is more than enough to configure comitup through the web page.

I had no issues running comitup along with Pihole.

davesteele, 2021-02-26:
@Anjum9694, thanks for the information.

closed 2021-04-10 by davesteele