davesteele/comitup #128: Connection is not available on the device wlan0 at this time

cdruet, 2020-11-24:
I have to install comitup on running Raspi's, on which I can't easily have access to the micro-SD card.

I've installed comitup according to https://github.com/davesteele/comitup/wiki/Installing-Comitup

I think...

First trouble was that pycairo wasn't available. I managed to install it. I'm stuck here :
`pi@raspberrypi:~ $ sudo /usr/sbin/comitup
Traceback (most recent call last):
  File "/usr/sbin/comitup", line 33, in <module>
    sys.exit(load_entry_point('comitup==1.13', 'console_scripts', 'comitup-cmd')())
  File "/usr/share/comitup/comitup/comitup.py", line 130, in main
    cdns.state_callback,
  File "/usr/share/comitup/comitup/statemgr.py", line 171, in init_state_mgr
    conf.ap_password
  File "/usr/share/comitup/comitup/states.py", line 332, in init_states
    set_state('HOTSPOT')
  File "/usr/share/comitup/comitup/states.py", line 265, in set_state
    state_info.start_fn()
  File "/usr/share/comitup/comitup/states.py", line 54, in wrapper
    returnvalue = fn(*args, **kwargs)
  File "/usr/share/comitup/comitup/states.py", line 106, in hotspot_start
    activate_connection(hs_ssid, 'HOTSPOT')
  File "/usr/share/comitup/comitup/states.py", line 281, in activate_connection
    path=path)
  File "/usr/share/comitup/comitup/nm.py", line 178, in activate_connection_by_ssid
    nm.NetworkManager.ActivateConnection(connection, device, path)
  File "<string>", line 6, in ActivateConnection
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.NetworkManager.UnknownConnection: Connection 'comitup-230-0000' is not available on the device wlan0 at this time.
`

davesteele, 2020-11-24:
Version? You should be using the package from the Comitup repository.

Is the connection file for "comitup-230-0000" in /etc/NetworkManager/system-connections?

cdruet, 2020-11-24:
A `comitup -v`would definitely help ;-)

I've added `deb http://davesteele.github.io/comitup/repo comitup main` to /etc/apt/sources.list and run `sudo apt update`and `sudo apt install comitup`

comitup-230-0000 is in system-connections

davesteele, 2020-11-24:
"dpkg --list comitup" or the "info" command in [comitup-cli](https://davesteele.github.io/comitup/man/comitup-cli.pdf).

Make sure [step 2](https://github.com/davesteele/comitup/wiki/Installing-Comitup) is done, and reboot if necessary.

You can debug by attempting the connection from NetworkManager ("sudo nmcli con up comitup-230-0000") and check in /var/log/ for NetworkManager messages.

cdruet, 2020-11-25:
I restarted the installation/configuration from stratch.

At step 2, shouldn't I also edit `/etc/dhcpcd.conf`to add just one line `denyinterfaces wlan0`?

I made it work using version 0.7-1 but it looks like there was not captive page by then. 

I upgraded to 1.13-1... and I ran into the same issue again... pycairo isn't automatically installed.

Then it ultimately worked ;-)

davesteele, 2020-11-25:
I'm confused. A dependency to "pycairo" has been in setup.py since v1.11, leading to a package dependency to "python3-cairo". Was there something like an "apt-get -f install" leading to it working?

Thanks for the input on the manual install instructions. I'll work it.

cdruet, 2020-11-25:
> I'm confused. A dependency to "pycairo" has been in setup.py since v1.11, leading to a package dependency to "python3-cairo". Was there something like an "apt-get -f install" leading to it working?
> 
> Thanks for the input on the manual install instructions. I'll work it.

No clue why. To make it work, I've had to install pip3 `sudo apt install python3-pip`and then `sudo pip3 install pycairo`

TwoD, 2020-12-06:
I had to also run 
```
sudo apt install python3-setuptools python3-dev pkg-config build-essential
sudo pip3 install wheel
sudo pip3 install pkgconfig # unsure about this one
sudo apt install libcairo2-dev
```
before `sudo pip3 install pycairo` would work on a Debian  based OSMC install.
Then `comitup.service` came up.

I also had to change the port in `/usr/share/comitup/web/comitupweb.py` for `comitup-web.service` to come up since Kodi is running on 80, would be great if that was configurable.

davesteele, 2020-12-06:
You don't want to move comitup-web off of port 80 - it is part of the strategy for captive portal detection. Also, if you do move it anyway, you'd need to update the DHCP config.

This is what the web_service configuration parameter is for.

closed 2021-04-23 by davesteele