davesteele/comitup #129: View SSID Password

Turbex, 2020-11-25:
Is there a build option to allow Comitup to display the complete entered SSID password like the "eyeball" icon seen in many places?  Entering a complicated password "blind" can be a bit of a pain if you have clumsy fingers!

davesteele, 2020-11-25:
That would be a function of the configuration of [UIKit](https://getuikit.com/) configuration in the [confirm.html](https://github.com/davesteele/comitup/blob/dd075e7d2312fa6a779c1cc8948dee3392faaf9b/web/templates/confirm.html) template.

I endeavor to respond to PRs promptly.

Turbex, 2020-11-25:
Brilliant - thanks!

Turbex, 2020-11-25:
Got it - uk-form-password as shown here: https://getuikit.com/v2/docs/form-password.html