davesteele/comitup #134: Making comitup compatible with a readonly system

Anjum9694, 2021-02-01:
Hello davesteele,

First issue of 2021 so a happy new year. I would like to make a fresh comitup installation compatible with this tutorial.
https://medium.com/swlh/make-your-raspberry-pi-file-system-read-only-raspbian-buster-c558694de79

The issue is that comitup does not start and there is no WiFi connection. I'm still able to SSH through ethernet though.
What can I do to make it compatible?

Thanks in advance

davesteele, 2021-02-01:
See #8. Most writes could be avoided with something like log2ram.

Anjum9694, 2021-02-02:
Thanks log2ram works fine along with comitup. Busybox-syslogd as mentioned in the tutorial wasn't playing nice with comitup.
Also I just found out about OverlayFS in raspi-config menu command. Comitup works fine with it. No problems now.

closed 2021-02-02 by Anjum9694

davesteele, 2021-02-02:
Yours is the first report of these working. Thanks.