davesteele/comitup #140: Connect two adapters simultaneously

sabertom, 2021-02-23:
Hi davidsteele,

First of all, awesome work, thank you!

Is it possible to use an additional external wifi adapter and let both adapters (internal and external) connect to the same network after the initial selection via AP? I want to connect to 2.4GHz and 5.5GHz simultaneously. If I set enable_appliance_mode=false the second adapter is just ignored. Thanks!

davesteele, 2021-02-23:
No. It's the router's responsibility to handle the bridging.

sabertom, 2021-02-23:
Thanks, understood. Though, when I use wpa_supplicant.conf to setup the wireless network, the internal adapter is linked via 5.5Ghz and the external (not capable of 5.5) uses 2.4GHz, both on the same network. How could I force the second adapter to join the network as well after the SSID is selected via AP in the first run?

davesteele, 2021-02-24:
The software does not have that capability.

closed 2021-02-24 by davesteele