davesteele/comitup #141: Unknown Comitup behaviour

Anjum9694, 2021-03-06:
I have my comitup working fine. However there is one thing I don't understand, it fails to connect to the AP exactly a week after boot. Only way to fix is by rebooting. It still creates a hotspot but the dhcp fails to assign an IP. I can still SSH through the hotspot, but the IP address has to be entered manually (10.41.0.1 gateway). This behavior is on a fresh installed comitup image, by the way.

Since I can access through the hotspot, what can I check to debug the behaviour? As a workaround I just made a crontab to restart comitup and dhcpd service every week.

davesteele, 2021-03-06:
I don't see this behavior.

DNS is provided by dnsmasq.

See https://github.com/davesteele/comitup/blob/master/CONTRIBUTING.md.

davesteele, 2021-03-07:
If the dhcp is failing to assign an IP, how can you ssh through the hotspot?

Are you connecting to an upstream network?

Is this actually an avahi failure?

Anjum9694, 2021-03-07:
Hi, After exactly a week comitup fails to connect to the AP and creates a HOTSPOT. When connected to the HOTSPOT my device is stuck on obtaining IP address (DHCP failure?). I manually enter 10.41.0.150 as the IP and 10.41.0.1 as the Gateway.

I have currently 2 crontabs that execute when this happens (which is always a week after boot), I am restarting these services comitup and dhcpd. I will post how it goes.

davesteele, 2021-03-07:
Ok.Note:
- Comitup manages a dnsmasq instance (not  the dnsmasq service) to provide a DHCP server and DNS service on the primary WiFi interface. A failed dnsmasq is a possible reason you can't get an IP address after the initial failure. This is usually caused by contention for one of the server sockets dnsmasq wants to control.
- Neither comitup nor dnsmasq should be involved in keeping the upstream AP connection up. That should be all NetworkManager, who manages a dhcp client.

The link you provide strongly suggests that restarting comitup will not fix the problem. Unless all the reporters are running Comitup, then NetworkManager is not responsible either. That sounds more like the WiFi device driver. Try a USB WiFi (though there are many adapters that [don't work](https://github.com/davesteele/comitup/wiki/FAQ))? Your cron job may have to reboot the Pi.

The bottom of this [comment]https://www.raspberrypi.org/forums/viewtopic.php?t=240612#p1471017) mentions that Pi-Hole rewrites the dnsmasq configuration file, meaning that it is independently using dnsmasq. This is a major opportunity for conflict with Comitup.

davesteele, 2021-03-07:
If you are using this for Pihole, and Comitup is just about headless setup, why not directly connect the Pi to the router? The Ethernet interface would be configured automatically.

davesteele, 2021-03-07:
Something you can look at now - how long is the DHCP lease time for the AP connection. Is it 7 days perchance?

[nmcli -f ipv4.method con show "\`nmcli -t -f NAME connection \`"](https://unix.stackexchange.com/questions/284550/check-my-dhcp-lease)

Anjum9694, 2021-03-07:
Ah.. I see. Network Manager handles the connection to AP. Uncommented the restart cronjobs for now. I changed IPv4 to Manual and entered a static IP in nmtui for the AP connection.

`nmcli -f ipv4.method con show "`nmcli -t -f NAME connection `"` returns
```
Error: AP-SSID
dhcp
comitup Pi-0000
static - no such connection profile.
```

Anjum9694, 2021-03-12:
"Solved" for now by switching to ethernet. If anyone else has encountered this and knows a solution please comment on this issue.

davesteele, 2021-05-01:
You [say](https://github.com/davesteele/comitup/issues/87#issuecomment-830574982) that swap fills after pretty much exactly one week. Do you know which process is taking the memory? I've got two days in on a system - "top" mode "M" doesn't show any candidate problem children.

Something like this has come up [before](https://github.com/davesteele/comitup/blob/b53cc09740b6164b916d202b54f8245e966ab630/comitup/wpa.py). I wonder if this is related. Perhaps you can see if the wpa_cli commands in that file help.

Anjum9694, 2021-05-01:
It may not be caused by comitup, it could be related to a Grafana datasource prometheus I installed leaking into memory. I raised this issue in the first place in case anyone else encounters it.

closed 2021-05-01 by Anjum9694