davesteele/comitup #142: Com It Up and Apache2

GerritRie, 2021-03-08:
Hey Dave amazing Project !!!

I have a little Problem. First of all everthing is working perfect.

But now is my problem. I build on on Pi Zero w. and apache2 runs on it cause after connect with comitup to my wifi i need a setup up interface for my programm.  So when i boot the RPI i cant use Comitup to connect. after connect to my local Wifi i cant accsess my Index.php on the raspbian. So i need to restart apache after succsesfull connection of wifi. can u maybe help me a bit out where to put the restart of apache ?

best regards !!!

davesteele, 2021-03-08:
See the web_service parameter in comitup.conf.

GerritRie, 2021-03-08:
So should me in should switch web_service: httpd.service to web_service: apache2?

davesteele, 2021-03-08:
That's not really a question for here. Use whatever service name "systemctl status" reports.

closed 2021-04-10 by davesteele