davesteele/comitup #145: How can I move the /usr/share/comitup/web directory

revantjhalani, 2021-03-13:
I need to move the web directory to a different location, seems like I'd only need to change the path for comitupweb.py or the web folder but I can't find what is executing the flask application at startup

davesteele, 2021-03-13:
https://github.com/davesteele/comitup/blob/debian/debian/comitup.comitup-web.service

The referenced sbin script uses the python pkg_resources module. That module has some secret sauce that looks in /usr/share/&lt;pgk&gt; for python modules.

Replace /usr/sbin/comitup-web with something that sets up the PYTHONPATH the way you want it.

revantjhalani, 2021-03-13:
I looked at the file and it seems to be a copy of the file of the same name in the /usr/share/comitup directory
I looked up some references for load_entry _point and it seems to only be adding an executable comitup-web to the path (I'm not well versed on how path variable setups work so I may be wrong) 
I see no reference to the web folder or any executions for the same in any file in the entirety of the raspberry pi files whatsoever

davesteele, 2021-03-13:
The reference to "web" you are looking for is in entry_points in setup.py.

revantjhalani, 2021-03-13:
I don't think I'm experienced enough to find these links and edit them to change the location of the directory.
I feel like I'll have a better shot editing the cron job that runs the flask app at every startup 
Could you tell me how the autostart implementation has been done? 
I don't see anything in the rc.local and it's been a while but I don't remember seeing any cron job either. 
This leaves me to think that you've initialized this as a service, I could disable the service and run the flask app in rc.local, 
that way it'll run as root too(or is there a comitup user i need to run the application as?) 

Does this sound like it'll work?

davesteele, 2021-03-13:
Comitup is auto-started via comitup.service. Comitup-web is also a systemd service, but it is managed by comitup. Comitup is rather insistent on retaining control.

closed 2021-04-10 by davesteele