davesteele/comitup #146: Any inbuilt way to change the hostname to the comitup-<nn> variable?

revantjhalani, 2021-03-13:
If there are multiple comitup devices on the same network, there's bound to be hostname conflicts. 
Is there any inbuilt way to also change the hostname to the comitup-<nn> variable?

davesteele, 2021-03-13:
if you are up to date, it is &lt;nnn&gt;. There is a .1% of 2 hosts having the same name.
Also, look at the comitup.conf file.

revantjhalani, 2021-03-13:
The conf only specifies how to add the hostname in the ssid, what I'd like to do is the opposite. 
I'd like to add the number <nnn> as a suffix to the hostname.
Also, I don't understand how the possibility of two devices having the same hostname is .1%? 
For instance, the prebuilt image you deliver has a default hostname of raspberrypi, if I were to flash 2 pis with that image and then connect them to the same network, won't there be a hostname problem right then

davesteele, 2021-03-13:
True, but they also advertise on mdns as comitup-&lt;nnn&gt;.local.

There is automatic deconfliction with the "raspberrypi" name. I have "raspberrypi" and "raspberrypi-2" on my network now. I've not looked into how that is managed.

This would be a decent addition to the Comitup Image. The package, though, should not be changing hostnames.

davesteele, 2021-03-20:
The latest comitup image sets the hostname to the AP name on first boot.

revantjhalani, 2021-04-04:
In the new image, I keep running across the error "unable to resolve host comitup-931: Name or service not known" while invoking any system commands.
I haven't had a chance to look at the hosts and hostname files, I'll try to recreate the issue in a while, but thought that this could just be the behavior at the first system startup since you're now changing the hostname at runtime. Is that the case?

davesteele, 2021-04-04:
I haven't seen this.

minutmorph, 2021-04-12:
Hi David - first of all thank you for this project - it has become my starting point for every project recently! I have also been receiving this issue on the new image too (2021-03-17) - I am able to recreate it when running anything preceded with sudo. 

eg. on first boot - my first action is to update the ap_name variable in /etc/comitup.conf (to "product-<nnnn>") and change the hostname to "device-$MACADDRESS" (for the exact reason mentioned above - to prevent conflict). Both actions require sudo, and doing this programmatically is difficult now on the new image. The line below gets the wlan0 mac address as a string and sets the hostname

```
# Change hostname to device-$MACADDRESS
pi@comitup-591:~ $ sudo hostname device-$(cat /sys/class/net/wlan0/address | sed -e 's/://g')
sudo: unable to resolve host comitup-591: Name or service not known
```

A quick google found me here:
https://www.globo.tech/learning-center/sudo-unable-to-resolve-host-explained/

And the apparent solution is to include a loopback in /etc/hosts to the given hostname
ie.

```
127.0.0.1             mynewhostname
```

I also found a way to change the /etc/hosts file programmatically - here's the full bash script below

``` set_hostname.sh
#!/bin/bash

# Set Hostname
echo "Setting hostname..."
NEWHOSTNAME=$(cat /sys/class/net/wlan0/address | sed -e 's/://g')
sudo hostname device-$NEWHOSTNAME

# Add loopback to /etc/hosts file
echo "Adding loopback to /etc/hosts file..."
echo "127.0.0.1               ghost-$NEWHOSTNAME" | sudo tee -a /etc/hosts

```

Some notable strange behaviour that might be related:
1. When in AP mode - you can't even achieve changing the /etc/hosts file - sudo will not resolve, however, when connected to a host network, sudo has the same warnings but manages to update the file, seemingly fixing the problem!
2. The hostname *always* reverts back to the original session "comitup-<nnn>" hostname - even if the new config specifies the ap_name to be device-<nnnn> and the hostname was changed before.
3. the ap_name variables such as <MMMMMMMMMMMM> and <ssss> don't appear to work - I'm not sure if this is related or is a separate issue.... the AP simply displays as if <M> and<s> are strings rather than variables

I hope this helps.

davesteele, 2021-04-12:
The persistent random number is stored in /var/lib/comitup/comitup.json. If it is set before comitup runs, it will not be overwritten.

The service that sets the host name runs exactly once, then self-deletes. In general, I don't want Comitup messing with hostnames.

The new ap_name variables currently require an "apt-get upgrade" to come alive.

davesteele, 2021-04-12:
> And the apparent solution is to include a loopback in /etc/hosts to the given hostname

A disk image with this change is uploading now.

closed 2021-04-12 by davesteele