davesteele/comitup #147: Multiple instances of Comitup running near each other

davidtraver, 2021-03-14:
Dear Mr. Steele,

I'm working on a project where I am using comitup to generate access points for cameras inside bluebird houses in a field (with solar power for the Raspberry Pi Zeros).  So, I have one access point named Camera 1, the second named Camera 2, and so on.  Using this approach, a naturalist can pick which camera she wishes to see by selecting the wifi name for that box, and then looking at the camera at port 81

Because Comitup access points use the default IP (10.41.0.1) there is a conflict between the AP when they are in close proximity.  I'd like to give unique default IP addresses to each AP via SSH before installation in the field.  However, I can't seem to find the file which propigates the IP address.  Could you please help me with this?

Many thanks,

Dave

davesteele, 2021-03-14:
So you are keeping all in hotspot mode. There will be no conflict with the 10.41.0.1 address - they are all on separate networks.

davidtraver, 2021-03-15:
I'd like to have the ability to select one AP over the other by giving them unique IP and SSIDs   To do this, how can I reliably change the SSID and IP for each comitup Raspberry Pi?

Thank you.

davesteele, 2021-03-15:
The SSID can be controlled in comitup.conf. The IP is hard coded, but I don't think that is a problem here, unless the client has multiple WiFi's connected to multiple SSIDs simultaneously.

davidtraver, 2021-03-15:
Thanks!

Dave

closed 2021-04-10 by davesteele