davesteele/comitup #148: Change the default IP

davidtraver, 2021-03-14:
Dear Mr. Steele,

I see I did not explain correctly.  When two or more hotspots with the same IP are operating near each other, it is not possible to select one over the other.  , I need, please, the means to change the SSID, which I can do /etc/comitup.conf  but also change the IP to something other than 10.41.0.1   For example I'd like to create 10.41.0.60 with an SSID of Camera-60.  Then I can access that camera at 10.41.0.60:81   At the same time, another hotspot would be at 10.41.0.50 with an SSID of Camera-50.  We could access that camera at 10.41.0.50:81

Thank you.

Dave

davesteele, 2021-03-15:
So you expect to connect to more than one ssid/camera at a time? Using multiple WiFi adapters? Otherwise, your only concern is with ssid collisions. If you only connect to one ssid at a time, who cares if the IP addresses overlap? Camera 60 is accessed at 10.41.0.1 port 81 when connected to the Camera-60 hotspot.

To answer your question, though, That IP address is hard coded in the comitup and comitup-web source, and a pair of configuration files.

davidtraver, 2021-03-15:
Yes, I want to use several AP at one time.  I see from file searches that it would require using ghex and such to edit multiple files.  So, I'll try a different system.  Nonetheless, thank you for a wonderful piece of software, and for your quick replies.  Be safe.

Dave

davesteele, 2021-03-15:
Can you describe this piece of gear the naturalists are using to connect to multiple hotspots at the same time?

closed 2021-04-23 by davesteele