davesteele/comitup #151: how to disconnect from a Wifi

dheim-pi, 2021-03-25:
Hi, is there a way to reset or disconnect from a wifi by comitup.

davesteele, 2021-03-25:
Delete the connection using comitup-cli or D-Bus. See the man pages for comitup-cli and comitup.

dheim-pi, 2021-03-25:
Many Thanks for the quick replay.  I was not aware about comitup-cli.

revantjhalani, 2021-03-26:
@davesteele  I tried using the ciu_client.ciu_delete(connection) method by importing the library and loading the connection variable through state. 
But that basically disconnects from wifi without the wifi coming back up again.
My preferred method of disconnection via a python script was to just run a subprocess and pass d and then q to comitup-cli command but that works once or twice and then just doesn't.
Is there a better way to disconnect? Or am I doing something wrong with the ciu_delete method?

davesteele, 2021-03-26:
Ciu_client and d do the same thing. If you just disconnect, comitup will just reconnect again.

closed 2021-04-04 by davesteele