davesteele/comitup #152: Nothing happend when i run comitup-cli

9acca9, 2021-03-26:
Hi.
First time i try to use your work.
Im connect to the raspberry pi zero w throug  ssh. I install comit-up. And try to run comitup-cli but nothing happend.
What could i try?
Thanks.

davesteele, 2021-03-27:
Comitup-cli will have to be restarted whenever comitup is restarted. It won't help here.

The line in question runs the command "nmcli device show", as root. Can you post the output of that command?

9acca9, 2021-03-27:
@davesteele 
HI.
Thanks.

I have this from the command. It seems related to my language (spanish).

```
GENERAL.DEVICE:                         wlan0
GENERAL.TYPE:                           wifi
GENERAL.HWADDR:                         F2:91:0B:50:3D:E7
GENERAL.MTU:                            1500
GENERAL.STATE:                          20 (no disponible)
GENERAL.CONNECTION:                     --
GENERAL.CON-PATH:                       --

GENERAL.DEVICE:                         lo
GENERAL.TYPE:                           loopback
GENERAL.HWADDR:                         00:00:00:00:00:00
GENERAL.MTU:                            65536
GENERAL.STATE:                          10 (sin gesti<F3>n)
GENERAL.CONNECTION:                     --
GENERAL.CON-PATH:                       --
IP4.ADDRESS[1]:                         127.0.0.1/8
IP4.GATEWAY:                            --
IP6.ADDRESS[1]:                         ::1/128
IP6.GATEWAY:                            --
IP6.ROUTE[1]:                           dst = ::1/128, nh = ::, mt = 256

```

In "general state" probably says "sin gestión" with the ó. How i can fix this?
Thanks for your time.

davesteele, 2021-03-27:
The special character is the problem, but it shouldn't be. The code should handle it.
This is just hardware checking code. I'm going to just tolerate the error (a458fbe). If you would like to try this out, edit the file at /usr/share/comitup/comitup/wificheck.py on your system.

9acca9, 2021-03-27:
> The special character is the problem, but it shouldn't be. The code should handle it.
> This is just hardware checking code. I'm going to just tolerate the error [...]

HI.Thanks. But, how i can edit the file. I mean what i have to write/comment out??

9acca9, 2021-03-27:
mmm, i just see this https://github.com/davesteele/comitup/wiki/Installing-Comitup
now is working!
Beatiful! Very useful!

davesteele, 2021-03-27:
If you want to add to the comitup web interface (available only in hotspot mode), it's one of the files in "templates".

davesteele, 2021-03-28:
> HI.Thanks. But, how i can edit the file. I mean what i have to write/comment out??

For anyone else wanting to make this change, just make the first line of wificheck.py device_nm_managed() "return None", with the text indented four spaces.

closed 2021-03-28 by davesteele