davesteele/comitup #154: Does not opening hotspot in comitup lite image

harshitjawla, 2021-03-27:
Hello. I run raspberry pi over local *ssh* through my *android* phone with a ssh app. I dont have any monitor. I used RaspiOS lite (original).

I have used several other methods for auto wifi-hotspot switch if one is unavailable. But failed in all, because during their setup all disconnects wifi and ssh session is terminated.

I recently booted comitup RaspiOS lite image (~450mb) on my Raspberry pi model B. But unfortunately, when I turn on Raspberry pi, its name does not show up in the list:
[image]
I just booted and powered on headlessly without(but hotspot not openning automatically), am I missing something? Please help

davesteele, 2021-03-27:
Have you gone through [CONTRIBUTING.md](https://github.com/davesteele/comitup/blob/master/CONTRIBUTING.md)?

harshitjawla, 2021-03-27:
Ooo so sorry.
I m crazy. I forgot to put sd card into raspberry after flashing.
Now everything is fine. Thankyou for this amazing solution package.
Please forgive me and Close this siht

davesteele, 2021-03-27:
That's a new one. Enjoy.

closed 2021-03-27 by davesteele