davesteele/comitup #160: data['sn']: _getserial()

mastin-zgz, 2021-04-14:
Hi!
I am trying to install the system on a Nano Pi R1 which is very similar to the RAspberry and has ubuntu xenial.

I've listed the repository, updated, and done an apt get install comitup, but it gave me this error.

Setting up comitup (1.19-1) ...
  File "/usr/share/comitup/comitup/config.py", line 129
    data['sn']: _getserial()
              ^
SyntaxError: invalid syntax

dpkg: error processing package comitup (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 comitup
E: Sub-process /usr/bin/dpkg returned an error code (1)

any ideas?

Thanks!

davesteele, 2021-04-15:
The failure is [confirmed](https://github.com/davesteele/comitup/runs/2348847614), but you are seeing it because Xenial is running Python 3.5, which is out of support.

closed 2021-04-15 by davesteele

davesteele, 2021-04-18:
See #167