davesteele/comitup #162: Comit-up is unable to connect to previously used WiFi location.

RamonBlandford, 2021-04-15:
Issue occurred when a Fuse in my home blew.  The router, the RPI (or both) restarted.
RPI went into WiFi hostpot mode and could not see previously used router connection.
Restarting RPI did no result in reconnection to home wifi.

The only way to fix this was to connect the RPI to my phone's WiFi tether, turn off tether and repeat connection process.  This time the Home wifi was visible in web list and could be used.

davesteele, 2021-04-15:
Is this #161?

davesteele, 2021-04-15:
Does comitup [v1.17](https://davesteele.github.io/comitup/deb/comitup_1.17-1_all.deb) work better?

davesteele, 2021-04-15:
Fixed in 1.20.

closed 2021-04-15 by davesteele

RamonBlandford, 2021-04-17:
Hmm, might be so!  I'll give the latest version of this fantastic utility a go and write back.
Peace

RamonBlandford, 2021-04-19:
Hi Dave, 
I can confirm that this bug still exists in manually installed version 1.3.1-1.
Does the fact that this was a manual installation rather than an image installation mean anything?

davesteele, 2021-04-20:
Very likely. See #163. And note that 1.3.1 is way old and has known problems. Upgrade.

RamonBlandford, 2021-04-20:
Hi Dave, 
So can I confirm that ver. 1.20. is newer than ver. 1.3.1?

davesteele, 2021-04-20:
[Yes](https://github.com/davesteele/comitup/blob/debian/debian/changelog)