davesteele/comitup #163: dnsmasq fails to use port 53

mastin-zgz, 2021-04-15:
Good morning!

Randomly, it stops me assigning ip in access point mode.
`dnsmasq: failed to create listening socket for port 53: Address already in use`
I have deactivated and prayed the systemd.resolved.
I have applied the "systemctl mask dnsmasq.service"
The error continues.
If I look at who uses port 53, I get this.

`udp        0      0 127.0.1.1:53            0.0.0.0:*                           0          22265       3056/dnsmasq
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           110        7432        486/avahi-daemon: r
udp6       0      0 :::5353                 :::*                                110        7433        486/avahi-daemon: r`

Is there any other option that may apply?

Thanks!

mastin-zgz, 2021-04-15:
As an added information, I have a cable connected to eth0 that provides internet to the machine. I don't know if that could be the cause.

davesteele, 2021-04-15:
"systemctl status" can tell you who launched the dnsmasq that is hogging the port.

eth0 is not a problem.

mastin-zgz, 2021-04-16:
Reinstaling the S.O solved the problem.

Thanks!

Ataraxiall, 2021-04-19:
Its happening again, and is the networkManager

[image]

i edit the  /etc/NetworkManager/NetworkManager.conf

and change

`[main]
plugins=ifupdown,keyfile,ofono
dns=dnsmasq

[ifupdown]
managed=false`

by
`[main]
plugins=ifupdown,keyfile,ofono
dns=8.8.8.8

[ifupdown]
managed=false`

It works but i dont know if is bad change this dns option in NetworkManager.conf...

davesteele, 2021-04-20:
The _NetworkManager.conf_ file in the Comitup image:

    [main]
    plugins=ifupdown,keyfile
    
    [ifupdown]
    managed=false

davesteele, 2021-04-20:
I've added this to the "installing' wiki page.

closed 2021-04-20 by davesteele