davesteele/comitup #164: Unable to connect when use password

Ataraxiall, 2021-04-16:
Hello everybody!
I'm testing the program and it works great, the only downside is that when I modify the /etc/comitup.conf and put a password, the computers do not connect.

My log 
`wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
AP-STA-POSSIBLE-PSK-MISMATCH fa:13:d1:c3:58:68`

/etc/comitup.conf passworrd
`ap_password: supersecretpassword`
My etc/NetworkManager/system-connections/
`[wifi-security]
key-mgmt=wpa-psk
psk=supersecretpassword`

Any idea?

Thanks!

davesteele, 2021-04-16:
I just tried it - it worked fine.

The log suggests you typed the password in wrong. Also note that it has to be at least 8 characters.

Just to be clear, this password is for the Comitup hotspot.

Ataraxiall, 2021-04-19:
i reinstalled wpa_suplicant package and it works.

The password was fine and had more than 8 characters, I don't know what could have been, but reinstalling it has been solved

Thanks!

closed 2021-04-19 by Ataraxiall