davesteele/comitup #169: Doesn't work when entering incorrect credentials to wifi

revantjhalani, 2021-04-23:
The AP doesn't come back up after entering incorrect credentials 
This is the error that comes up when I enter incorrect wifi details:

"comitup[446]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
File "/usr/local/lib/python3.7/dist-packages/dbus/connection.py", line 232, i$Apr 23 22:00:33 Comitup-56 comitup[446]:     self._handler(*args, **kwargs)
File "/usr/share/comitup/comitup/nmmon.py", line 111, in any_changed_state
mdns.add_hosts(dns_names)
File "/usr/share/comitup/comitup/mdns.py", line 153, in add_hosts
group.Commit()
File "/usr/local/lib/python3.7/dist-packages/dbus/proxies.py", line 147, in _$Apr 23 22:00:33 Comitup-56 comitup[446]:     **keywords)
File "/usr/local/lib/python3.7/dist-packages/dbus/connection.py", line 653, i$Apr 23 22:00:33 Comitup-56 comitup[446]:     message, timeout)
dbus.exceptions.DBusException: org.freedesktop.Avahi.IsEmptyError: Is empty"

I built this image using the pi-gen fork for comitup
avahi-daemon is installed

davesteele, 2021-04-24:
Dup of #106

closed 2021-04-24 by davesteele

revantjhalani, 2021-04-24:
@davesteele How is that even remotely relevant to what my issue is, I don't understand. 
I've tried compiling the image on Ubuntu xenial (32 bits), Debian buster 32 bits and Ubuntu 64 bits, every single one of them has the same error. 
And the compilation is without any errors
Could you please point me in a direction if you've faced the same issue or if you know what it might be

revantjhalani, 2021-04-25:
I've reproduced this issue in the base image as well now, 
This happens after updating system time