davesteele/comitup #170: New install failing to update beyond "Setting up comitup (1.21-1)"

b-bilko, 2021-04-24:
Brand new install on a new Rpi3, using image_2021-04-12-Comitup-lite 

Running sudo apt-get upgrade, runs though normally and then gets to Setting up comitup (1.21-1) ... before locking up. 

Left for over an hour before rebooting and now won't reconnect. 

I've reflashed twice and fails at the same spot.

b-bilko, 2021-04-24:
I tried a reboot post update and the upgrade command worked. 

Not sure I can reproduce now, but happy to try!

davesteele, 2021-04-25:
Interesting. Upgrading can interrupt WiFi (which I guess puts the child upgrade session at risk), but, for me, the SSH session comes back in a minute or so.

[Unattended-upgrade](https://packages.debian.org/sid/unattended-upgrades)?

"sudo apt-get upgrade -y &"?

Or just do this for upgrading Comitup - "sudo apt-get install -y comitup &"

Is your ssh originating from Windows?

b-bilko, 2021-04-26:
I'll have a go on a new SD card, as I got it working with a reboot. 

I left it for 2+ hours at one point and it didn't recover, and post reboot it was a matter of seconds to proceed beyond the Setting up comitup (1.21-1) stage.

b-bilko, 2021-04-26:
I had to reinstall for another reason, and I've been able to reproduce the same issue. 

Session is from Windows, same in both Putty and Kitty.

b-bilko, 2021-04-26:
I rebooted and then tried: 

sudo apt-get install -y comitup &
[1] 797
Reading package lists... Done
Building dependency tree
Reading state information... Done
comitup is already the newest version (1.21-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

davesteele, 2021-04-27:
Try 1.22-1.

closed 2021-05-09 by davesteele