davesteele/comitup #171: Comitup stops working after updating system time

revantjhalani, 2021-04-25:
When trying to update on a fresh image (https://steele.debian.net/comitup/image_2021-04-12-Comitup-lite.zip) a system time issue is encountered, leading me to update the system time using "date -s"
After updating the date, the issue encountered in #169 occurs 
`
"comitup[446]: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
File "/usr/local/lib/python3.7/dist-packages/dbus/connection.py", line 232, i$Apr 23 22:00:33 Comitup-56 comitup[446]:     self._handler(*args, **kwargs)
File "/usr/share/comitup/comitup/nmmon.py", line 111, in any_changed_state
mdns.add_hosts(dns_names)
File "/usr/share/comitup/comitup/mdns.py", line 153, in add_hosts
group.Commit()
File "/usr/local/lib/python3.7/dist-packages/dbus/proxies.py", line 147, in _$Apr 23 22:00:33 Comitup-56 comitup[446]:     **keywords)
File "/usr/local/lib/python3.7/dist-packages/dbus/connection.py", line 653, i$Apr 23 22:00:33 Comitup-56 comitup[446]:     message, timeout)
dbus.exceptions.DBusException: org.freedesktop.Avahi.IsEmptyError: Is empty"
`

revantjhalani, 2021-04-25:
I did a bit more testing and it seems like the crash has nothing to do with time as well. 
I downgraded the files manually to version 1.19 and the system now works. 
Haven't tested 1.20 out yet, but the issue is certainly specific to version 1.21

revantjhalani, 2021-04-25:
Specifying the version of comitup to install doesn't work with apt `apt install comitup=1.19`  or `apt install comitup=1.19-1` both return that the version wasn't found. So I doubt that specifying the version in the pi-gen fork would work either.
Is there any way for me to specify the version in the pi-gen build?

davesteele, 2021-04-27:
Does 1.22-1 work better?

revantjhalani, 2021-05-02:
Just tried the new version today. 
The module doesn't crash but now the logs are just populated with the same error rather than just once like before. 
Nothing weird in daemon.log now, which is weird 
And the WiFi still fails to spawn. 
One thing I'd like to add, would it be possible to not push the untested releases to the apt mainline, it's bound to break the functionality for everyone who does a system upgrade. 
The last working version I've seen is 1.19, haven't tested 1.20, but both the releases after that don't work.

davesteele, 2021-05-04:
I'll have a word with QA.

The error is a side effect of a more aggressive strategy for populating ZeroConf. It's not necessarily a problem for the list to be occasionally empty. There are not daemon.log entries  anymore because I catch the exception and avoid the crash (which was the problem).  I do still log it as an exception in comitup.log. 

You say it doesn't work. Is there bad behavior beyond the Traceback log entry I added?