davesteele/comitup #172: Erratic connection after pairing to a second wifi network

breboulet, 2021-04-26:
Hi Dave,

Here are the steps to reproduce the weird behavior I ran into:
 - Start RPi4 from a clean comitup image
 - Comitup Hotspot shows up -> Connect to Wifi-Network-1 -> Success -> everything's happy, even across reboots
 - Turn Off RPi4
 - Turn Off Wifi-Network-1 (my Mac Mini's w/ Internet Sharing On)
 - Turn On Wifi-Network-2 (my usual Wifi Router)
 - Turn On RPi4
 - Comitup Hotspot shows up -> Connect to Wifi-Network-2 -> Success (or what looks like it)
 - 'ping google.com' produces erratic/random response times (sometimes 30ms, sometimes 6000ms+) <-- this doesn't happen on first pairing
 - On reboot re-connection seems slow
 - If I switch the networks again (Turn Off Wifi-Network-2 and Turn On Wifi-Network-1), hotspot shows up, even if repairing, connection fails
 
I couldn't pin down what the issue was, nor find any relevant logs, but something fishy is going on with the network after that second pairing. CPU usage seems normal. That's a tricky one...

Hopefully you'll be able to reproduce the issue and understand what's going on a little better than myself!!

And thanks again for that wonderful project ;)

Best,
Bastien

davesteele, 2021-04-26:
See [CONTRIBUTING](https://github.com/davesteele/comitup/blob/main/CONTRIBUTING.md) for logging and reporting notes.

breboulet, 2021-04-27:
Hi Dave,

Here's my test environment:
 - Stock Raspberry Pi 4 2GB with built-in wifi (no secondary dongle)
 - Comitup (non-lite image) generated via stock (no code change) https://github.com/davesteele/pi-gen with comitup v1.21-1 (see [2021-04-26-Comitup.info.txt](https://github.com/davesteele/comitup/files/6380999/2021-04-26-Comitup.info.txt) for full image info)
 - Two Wifi Routers (WifiNetwork1 is a MacMini sharing its internet connection - eth to wifi, WifiNetwork2 is the real router)
 - I had to plug the RPi to a monitor (hence the non-lite image) since I couldn't ssh into the broken RPi because of the network issues

Steps to reproduce:
 - Turn On WifiNetwork1
 - Turn On RPi
 - Comitup Hotspot shows up
 - Pair to WifiNetwork1 successful
 - Turn Off WifiNetwork1
 - Comitup Hotspot shows up
 - Pair to WifiNetwork2 unsuccessful/partial (couldn't even get to the erratic connection this time, it simply wouldn't connect to WifiNetwork2, Hotspot not showing up). Connection seems to be added to nmcli c though.
 - Reboot, same behavior: can't connect to WifiNetwork2, no hotspot
 - I tried comitup-cli: it shows the old connection WifiNetwork1 has the "connecting" one, even if I try to manually delete it (via comitup-cli) it wouldn't delete it. 'nmcli c' still shows both connections. I have the feeling that it would keep trying to use the very first connection no matter what.

Note that _individually_ (not in sequence, from scratch) connecting to WifiNetwork1 or WifiNetwork2 works without any issue.

Here are all the log files I collected while running the above steps:
[comitup.log](https://github.com/davesteele/comitup/files/6381000/comitup.log)
[comitup-web.log](https://github.com/davesteele/comitup/files/6381001/comitup-web.log)
[daemon.log](https://github.com/davesteele/comitup/files/6381003/daemon.log)
[syslog.txt](https://github.com/davesteele/comitup/files/6381004/syslog.txt)

Hope this helps.

Best,
Bastien

davesteele, 2021-04-27:
The daemon log indicates it is a known problem in recent versions. Upgrade or install 1.22-1.

breboulet, 2021-04-27:
I confirm the problem is resolved with 1.22-1
Thank you so much!
Best,
Bastien

breboulet, 2021-05-04:
Hi Dave,

For some weird unknown reason the problem is back there, after reboots it connects to wifi, but shorty after connection becomes unstable until it drops completely.
Using comitup-1.22-1.

See the new logs attached:
[syslog.txt](https://github.com/davesteele/comitup/files/6418655/syslog.txt)
[comitup.log](https://github.com/davesteele/comitup/files/6418652/comitup.log)
[comitup-web.log](https://github.com/davesteele/comitup/files/6418653/comitup-web.log)
[daemon.log](https://github.com/davesteele/comitup/files/6418654/daemon.log)

breboulet, 2021-05-07:
Here are more logs hat I gathered after letting the Pi run for a while. I thought they might be helpful. If you need anything else, please let me know ;)
[syslog.txt](https://github.com/davesteele/comitup/files/6443090/syslog.txt)
[daemon.log](https://github.com/davesteele/comitup/files/6443092/daemon.log)
[comitup-web.log](https://github.com/davesteele/comitup/files/6443093/comitup-web.log)
[comitup.log](https://github.com/davesteele/comitup/files/6443094/comitup.log)

closed 2021-05-08 by davesteele

breboulet, 2021-05-19:
Hi Dave,

Unfortunately I think the issue is back with comitup 1.25-1.
Note: I have the following comitup.conf set: `enable_appliance_mode: false`

I also noticed that, even when the connection works, the Pi has always 2 different IP addresses on the wlan0 interface, is that normal? (seen across multiple versions of comitup)

Best,
Bastien

See logs attached:
[comitup.log](https://github.com/davesteele/comitup/files/6505428/comitup.log)
[comitup-web.log](https://github.com/davesteele/comitup/files/6505429/comitup-web.log)
[daemon.log](https://github.com/davesteele/comitup/files/6505430/daemon.log)
[syslog.txt](https://github.com/davesteele/comitup/files/6505431/syslog.txt)

breboulet, 2021-05-19:
The double IP problem seems to be caused by dhcp**c**d.service
If I mask it, I then see a single IP address after reboot. Unmask it and double IP is back.

davesteele, 2021-05-19:
The comitup image has dnsmasq.service, systemd-resolved.service, and dhcpd.service masked.

breboulet, 2021-05-19:
It does have dhcpd.service masked, but it doesn’t have dhcpCd.service masked.

davesteele, 2021-05-19:
Concerning the double IP, I'm guessing you have a dhcp service connected to the hotspot? That could cause problems regardless of Comitup's behavior.

davesteele, 2021-05-19:
V1.27-1 is in the archive, but not the repo (yet).

breboulet, 2021-05-20:
Thanks Dave, I’ll give it a try when released in the repo and let you know ;)

breboulet, 2021-05-22:
I generated a new image with 1.28-1 and even though comitup-cli states it's in HOTSPOT mode and lists the nearby wifis, the actual hotspot does not show up on my iPhone's wifi network list.

davesteele, 2021-05-22:
1.29-1 released.

breboulet, 2021-05-22:
It works!!

commit e67bfe88 (2021-04-27): Release 1.22-1, fixing crash on hotspot startup

Should fix #172, #171, and #170
commit b2fb3280 (2021-05-08): Manage state changes better

The connecting_fail state change back to CONNECTING (for multiple
entries in the conn_list) was not happening. Also, set_state_to()
was being called multiple times from the same state. Timeout issue?

Fixes #172
commit 74928e4e (2021-05-08): Manage state changes better

The connecting_fail state change back to CONNECTING (for multiple
entries in the conn_list) was not happening. Also, set_state_to()
was being called multiple times from the same state. Timeout issue?

Fixes #172
commit bc05390c (2021-05-19): Don't re-initialize conn list in connecting start

Recent changes mistakenly caused the connection list to be reinitialized
in connecting_start(). That list is supposed to be depleted as
additional connection opportunities are attempted, causing the system to
be unable to connect to a second defined WiFi connection.

Fixes #172 (again)