davesteele/comitup #173: Add "factory reset" API to comitup-cli

breboulet, 2021-04-27:
Hi Dave,

While struggling with connection issues, I thought that this feature could be quite necessary for mitigating those kind of problems when you don't have a laptop available for troubleshooting.

That would allow the user/dev to plug it to a "long press" kind of button on the Pi. Comitup would then wipe all previous connections or configs it previously modified/created and go back to hotspot mode.

Pretty much all IOTs have that kind of reset button for when things go south, or when you just want to wipe your credentials out of it when handing it over to someone else.

Best,
Bastien

Ataraxiall, 2021-06-10:
Im using node red software and y can execute one line command. Exist a command to remove all saved networks?remove a file o something? I cant use cli because the exec node only cant send one line and if cli ask me a option, y cant response.

davesteele, 2021-06-10:
You'll want to delete the defined NetworkManager connections.

davesteele, 2021-06-19:
The "nuke" hw/sw-triggered feature is in v1.30-1.

closed 2021-06-19 by davesteele