davesteele/comitup #182: dnsmasq SIGTERM

sistemicorp, 2021-07-10:
After a reboot, `comitup` doesn't complete network connection to a known WiFi network.

Initially the WiFi network is unknown and `comitup` successfully enters hotspot and I can successfully navigate to the setup page and select the network and password and successfully connect to the network.

However, one reboot, although the RPi has connected to the WiFi network, it does not call the external callback.  I think this is showing the error,

    pi@p1125-f85e:~ $ sudo systemctl status comitup
    - comitup.service - Comitup Wifi Management
       Loaded: loaded (/lib/systemd/system/comitup.service; disabled; vendor preset: enabled)
       Active: active (running) since Sat 2021-07-10 18:51:56 EDT; 2min 3s ago
         Docs: man:comitup(8)
     Main PID: 2399 (comitup)
        Tasks: 1 (limit: 3727)
       CGroup: /system.slice/comitup.service
               └─2399 /usr/bin/python3 /usr/sbin/comitup
    
    Jul 10 18:51:55 p1125-f85e systemd[1]: Starting Comitup Wifi Management...
    Jul 10 18:51:56 p1125-f85e systemd[1]: Started Comitup Wifi Management.
    Jul 10 18:51:56 p1125-f85e dnsmasq[2438]: started, version 2.80 cachesize 150
    Jul 10 18:51:56 p1125-f85e dnsmasq[2438]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile
    Jul 10 18:51:56 p1125-f85e dnsmasq-dhcp[2438]: DHCP, IP range 10.41.0.100 -- 10.41.0.150, lease time 10m
    Jul 10 18:51:56 p1125-f85e dnsmasq[2438]: reading /etc/resolv.conf
    Jul 10 18:51:56 p1125-f85e dnsmasq[2438]: using nameserver 192.168.86.1#53
    Jul 10 18:51:56 p1125-f85e dnsmasq[2438]: read /etc/hosts - 4 addresses
    Jul 10 18:51:56 p1125-f85e dnsmasq[2438]: exiting on receipt of SIGTERM

`dnsmasq` is exiting.  I think this is the issue...

comitup log:

    2021-07-10 18:51:55,885 - comitup - INFO - Starting comitup
    2021-07-10 18:51:56,169 - comitup - INFO - Setting state to HOTSPOT
    2021-07-10 18:51:56,177 - comitup - INFO - Activating hotspot
    2021-07-10 18:51:56,499 - comitup - INFO - Setting state to CONNECTING
    2021-07-10 18:51:56,659 - comitup - INFO - Setting state to CONNECTED

I confirmed all the setup instructions from here: https://github.com/davesteele/comitup/wiki/Installing-Comitup

Connecting after a reboot was working, perhaps 1 month ago I tested it... But something has changed to break it and I haven't been able to figure it out... where to look... 

    pi@p1125-f85e:~ $ sudo cat /etc/hosts
    127.0.0.1	p1125-f85e
    ::1		localhost ip6-localhost ip6-loopback
    ff02::1		ip6-allnodes
    ff02::2		ip6-allrouters

The problem is occurring on the current version I have installed, AND it also happens if I update to the latest version.

When the WiFi network is not known, and the connection is made the first time, `dnsmasq` does not SIGTERM.

davesteele, 2021-07-11:
Assuming you are not using the Comitup Image, make sure that the dnsmasq service is at least disabled.

https://github.com/davesteele/comitup/blob/d8b170e/comitup/comitup.py#L59

sistemicorp, 2021-07-11:
Correct, I am not using the comitup image.
I see no log warnings in the comitup log.
Yes, I believe `dnsmasq.service` is disabled... here is what is running...

    pi@p1125-f85e:~ $ sudo systemctl --type=service
    UNIT                                                        LOAD   ACTIVE SUB     DESCRIPTION                                                       
    alsa-restore.service                                        loaded active exited  Save/Restore Sound Card State                                     
    alsa-state.service                                          loaded active running Manage Sound Card State (restore and store)                       
    avahi-daemon.service                                        loaded active running Avahi mDNS/DNS-SD Stack                                           
    bluetooth.service                                           loaded active running Bluetooth service                                                 
    bthelper@hci0.service                                       loaded failed failed  Raspberry Pi bluetooth helper                                     
    comitup.service                                             loaded active running Comitup Wifi Management                                           
    console-setup.service                                       loaded active exited  Set console font and keymap                                       
    containerd.service                                          loaded active running containerd container runtime                                      
    cron.service                                                loaded active running Regular background program processing daemon                      
    cups-browsed.service                                        loaded active running Make remote CUPS printers available locally                       
    cups.service                                                loaded active running CUPS Scheduler                                                    
    dbus.service                                                loaded active running D-Bus System Message Bus                                          
    dhcpcd.service                                              loaded active running dhcpcd on all interfaces                                          
    docker.service                                              loaded active running Docker Application Container Engine                               
    dphys-swapfile.service                                      loaded active exited  dphys-swapfile - set up, mount/unmount, and delete a swap file    
    fake-hwclock.service                                        loaded active exited  Restore / save the current clock                                  
    getty@tty1.service                                          loaded active running Getty on tty1                                                     
    gldriver-test.service                                       loaded active exited  Check for v3d driver                                              
    hciuart.service                                             loaded active running Configure Bluetooth Modems connected by UART                      
    ifupdown-pre.service                                        loaded active exited  Helper to synchronize boot up for ifupdown                        
    keyboard-setup.service                                      loaded active exited  Set the console keyboard layout                                   
    kmod-static-nodes.service                                   loaded active exited  Create list of required static device nodes for the current kernel
    lightdm.service                                             loaded active running Light Display Manager                                             
    ModemManager.service                                        loaded active running Modem Manager                                                     
    networking.service                                          loaded active exited  Raise network interfaces                                          
    NetworkManager-wait-online.service                          loaded active exited  Network Manager Wait Online                                       
    NetworkManager.service                                      loaded active running Network Manager                                                   
    polkit.service                                              loaded active running Authorization Manager                                             
    raspi-config.service                                        loaded active exited  LSB: Switch to ondemand cpu governor (unless shift key is pressed)
    rc-local.service                                            loaded active running /etc/rc.local Compatibility                                       
    rng-tools.service                                           loaded active running rng-tools.service                                                 
    rpi-eeprom-update.service                                   loaded active exited  Check for Raspberry Pi EEPROM updates                             
    rsyslog.service                                             loaded active running System Logging Service                                            
[... cut ...]
1) When the WiFi is unknown, "it works", there is no SIGTERM message from `dnsmasq` and the external callback is called.
2) On reboot, when it fails, the WiFi is connected to the network... but there is the SIGTERM message and the external callback is not called.
3) If I plug in `eth0`, which triggers stopping of `comitup` service (and wifi via `nmcli radio wifi off`), and then I unplug `eth0`, which triggers starting `comitup` "it works" (no SIGTERM, and external callback is called).  But just restarting `comitup` doesn't work.  Its like it needs the `eth0` connection to happen.

davesteele, 2021-07-11:
> if I disable that service, it doesn't fix the problem, and I lose `eth0` function.

I'm not seeing that problem. My eth0 is managed by NetworkManager.

> If I plug in eth0, which triggers stopping of comitup service (and wifi via nmcli radio wifi off), and then I unplug eth0, which triggers starting comitup "it works" (no SIGTERM, and external callback is called).

That sounds like your modifications are involved with the issue.

sistemicorp, 2021-07-11:
Indeed the issue was mine.  The root problem was my app was missing external callback updates as it was busy doing something else, the callback status updates were lost on my side.

It seems the `dnsmasq` SIGTERM is an expected, or normal occurrence.

closed 2021-07-11 by sistemicorp