davesteele/comitup #184: Connection activation failed: (7) Secrets were required, but not provided

d8ahazard, 2021-08-10:
The error is in the title. 

When creating a "new" connection with a device that's already been connected to a Google Wifi router, this error is thrown after going through the regular steps in the captive portal.

What *seems* to be happening is that my router knows this is a device that was already connected to it, and so is requesting a secret, which in this case would possibly be the BSSID of the point it wants to use? 

What's making it worse is that I also have multiple APs all using the same SSID, so I'd imagine that causes some confusion when telling the web UI what AP to use by name only.

Would it be possible to maybe add some kind of "advanced options" checkbox to the UI, and then have a more fine-tuned method of selecting the APN and/or providing a BSSID? Still poking at this to see if I can find more issues, but the message that shows up in syslog from the network manager is the one I posted. I can see the connection flow start and stop at that message, after which point network manager deletes the connection and re-starts the portal.

d8ahazard, 2021-08-10:
Following random guides, I tried to disable mac address randomization for NetworkManager. Still no success on connecting, but syslog shows the following:

`
Aug 10 16:20:38 glimmr NetworkManager[1851]: <info>  [1628630438.5543] Config: added 'key_mgmt' value 'WPA-PSK WPA-PSK-SHA256'
Aug 10 16:20:38 glimmr NetworkManager[1851]: <info>  [1628630438.5543] Config: added 'auth_alg' value 'OPEN'
Aug 10 16:20:38 glimmr NetworkManager[1851]: <info>  [1628630438.5543] Config: added 'psk' value '<hidden>'
Aug 10 16:20:41 glimmr wpa_supplicant[356]: wlan0: Trying to associate with SSID 'digitalhigh'
Aug 10 16:20:41 glimmr NetworkManager[1851]: <info>  [1628630441.0252] device (wlan0): supplicant interface state: disconnected -> associating
Aug 10 16:20:43 glimmr wpa_supplicant[356]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Aug 10 16:20:43 glimmr NetworkManager[1851]: <info>  [1628630443.7719] device (wlan0): supplicant interface state: associating -> disconnected
Aug 10 16:20:43 glimmr NetworkManager[1851]: <info>  [1628630443.8717] device (wlan0): supplicant interface state: disconnected -> scanning
Aug 10 16:20:46 glimmr wpa_supplicant[356]: wlan0: Trying to associate with SSID 'digitalhigh'
Aug 10 16:20:46 glimmr NetworkManager[1851]: <info>  [1628630446.4375] device (wlan0): supplicant interface state: scanning -> associating
Aug 10 16:20:48 glimmr systemd[1]: NetworkManager-dispatcher.service: Succeeded.
Aug 10 16:20:48 glimmr wpa_supplicant[356]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Aug 10 16:20:48 glimmr NetworkManager[1851]: <info>  [1628630448.9933] device (wlan0): supplicant interface state: associating -> disconnected
Aug 10 16:20:49 glimmr NetworkManager[1851]: <info>  [1628630449.0936] device (wlan0): supplicant interface state: disconnected -> scanning
Aug 10 16:20:51 glimmr wpa_supplicant[356]: wlan0: Trying to associate with SSID 'digitalhigh'
Aug 10 16:20:51 glimmr NetworkManager[1851]: <info>  [1628630451.6552] device (wlan0): supplicant interface state: scanning -> associating
Aug 10 16:20:54 glimmr wpa_supplicant[356]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Aug 10 16:20:54 glimmr NetworkManager[1851]: <info>  [1628630454.2719] device (wlan0): supplicant interface state: associating -> disconnected
Aug 10 16:20:54 glimmr NetworkManager[1851]: <info>  [1628630454.3721] device (wlan0): supplicant interface state: disconnected -> scanning
Aug 10 16:20:56 glimmr wpa_supplicant[356]: wlan0: Trying to associate with SSID 'digitalhigh'
Aug 10 16:20:56 glimmr NetworkManager[1851]: <info>  [1628630456.9350] device (wlan0): supplicant interface state: scanning -> associating
Aug 10 16:20:59 glimmr wpa_supplicant[356]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Aug 10 16:20:59 glimmr NetworkManager[1851]: <info>  [1628630459.5384] device (wlan0): supplicant interface state: associating -> disconnected
Aug 10 16:20:59 glimmr NetworkManager[1851]: <info>  [1628630459.6382] device (wlan0): supplicant interface state: disconnected -> scanning
Aug 10 16:21:02 glimmr wpa_supplicant[356]: wlan0: Trying to associate with SSID 'digitalhigh'
Aug 10 16:21:02 glimmr NetworkManager[1851]: <info>  [1628630462.2047] device (wlan0): supplicant interface state: scanning -> associating
Aug 10 16:21:03 glimmr NetworkManager[1851]: <warn>  [1628630463.8285] device (wlan0): Activation: (wifi) association took too long
Aug 10 16:21:03 glimmr NetworkManager[1851]: <info>  [1628630463.8293] device (wlan0): state change: config -> need-auth (reason 'none', sys-iface-state: 'managed')
Aug 10 16:21:03 glimmr wpa_supplicant[356]: wlan0: CTRL-EVENT-DISCONNECTED bssid=f0:72:ea:0f:ac:62 reason=3 locally_generated=1
Aug 10 16:21:03 glimmr NetworkManager[1851]: <warn>  [1628630463.8375] device (wlan0): Activation: (wifi) asking for new secrets
Aug 10 16:21:03 glimmr NetworkManager[1851]: <warn>  [1628630463.8381] sup-iface[0xd27488,wlan0]: connection disconnected (reason -3)
Aug 10 16:21:03 glimmr NetworkManager[1851]: <warn>  [1628630463.8407] device (wlan0): No agents were available for this request.
Aug 10 16:21:03 glimmr NetworkManager[1851]: <info>  [1628630463.8409] device (wlan0): state change: need-auth -> failed (reason 'no-secrets', sys-iface-state: 'managed')
Aug 10 16:21:03 glimmr wpa_supplicant[356]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD
Aug 10 16:21:03 glimmr NetworkManager[1851]: <warn>  [1628630463.8496] device (wlan0): Activation: failed for connection 'digitalhigh'
Aug 10 16:21:03 glimmr NetworkManager[1851]: <info>  [1628630463.8634] device (wlan0): supplicant interface state: associating -> disconnected
Aug 10 16:21:03 glimmr NetworkManager[1851]: <warn>  [1628630463.8654] sup-iface[0xd27488,wlan0]: connection disconnected (reason -3)
Aug 10 16:21:03 glimmr NetworkManager[1851]: <info>  [1628630463.8678] device (wlan0): state change: failed -> disconnected (reason 'none', sys-iface-state: 'managed')
Aug 10 16:21:04 glimmr NetworkManager[1851]: <info>  [1628630464.2217] audit: op="connection-delete" uuid="c19ca34d-0028-4d4f-a6d8-114ec7664243" name="digitalhigh" pid=2006 uid=0 result="success"

`

davesteele, 2021-08-13:
Comitup assumes that connections with the same SSID have the same configuration, including password.