davesteele/comitup #185: Error in v1.31-1: Exception encountered adding avahi record

john-h-watson, 2021-08-18:
I've updated from v1.29-1 to v1.31-1 yesterday (using apt-get update and apt-get upgrade) and wasn't able to get comitup running properly. The Raspberry would connect to the network, but comitup would always throw an error (see below) and stop, e.g. not starting my web service again.

 I tried to check for conflicts dnsmasq (DHCP), wpa_supplicant, and resolved (DNS) services but couldn't find any. The log would always end with the following:

```
2021-08-18 07:16:31,864 - comitup - INFO - Setting state to HOTSPOT
2021-08-18 07:16:31,936 - comitup - DEBUG - State call - hotspot-start
2021-08-18 07:16:31,937 - comitup - INFO - Activating hotspot
2021-08-18 07:16:32,287 - comitup - DEBUG - add_hosts: lo, 127.0.0.1
2021-08-18 07:16:32,449 - comitup - DEBUG - add_hosts: wlan0, 192.168.0.206
2021-08-18 07:16:32,486 - comitup - DEBUG - Add A record julsprint.local-2-192.168.0.206
2021-08-18 07:16:32,535 - comitup - DEBUG - Add AAAA record julsprint.local-2-2a02:8388:301:d280:8ec6:4373:3c5b:5784
2021-08-18 07:16:32,561 - comitup - ERROR - Exception encountered adding avahi record
```

I just downgraded to v1.30-1 and everything works just wonderful (Except running my own callback script, that's why I updated in the first place – but that's another question).

Thank you so much for your work!

davesteele, 2021-08-18:
Is IPv6 enabled on the system/wifi interfaces?

davesteele, 2021-08-18:
Sometimes, running comitup from the command line can give you a better look at the error.

    $ sudo systemctl stop comitup
    $ sudo comitup

davesteele, 2021-08-18:
... which is hidden in this case because of the [catch](https://github.com/davesteele/comitup/blob/f82143d764a0cfc27f075260d4ec8a34a4004c0f/comitup/mdns.py#L218) in mdns.py.

You could try changing this temporarily to a check for, say, KeyboardInterrupt.

davesteele, 2021-10-17:
Exceptions in mdns have been addressed. Delaying before resolving.

fc0987923939e07f04653a3476b151a469bd4408

2ba20648ca90e2b7e570ca4ac663447f5758e62f

b623639b848035a22bcc9bd18bd35e7ada172aba

closed 2022-02-21 by davesteele