davesteele/comitup #191: Some errors (Python??)

Ataraxiall, 2021-09-29:
Hello there,
I am having several problems, comitup does not work for me, it tells me that it cannot connect to the network, if I do
`sudo systemctl stop comitup`
and 
`sudo comitup`

this is the error

`Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nmmon.py", line 144, in reset_mdns
    mdns.add_hosts(dns_names)
  File "/usr/share/comitup/comitup/mdns.py", line 223, in add_hosts
    group.Commit()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.Avahi.IsEmptyError: Is empty
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nmmon.py", line 144, in reset_mdns
    mdns.add_hosts(dns_names)
  File "/usr/share/comitup/comitup/mdns.py", line 223, in add_hosts
    group.Commit()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.Avahi.IsEmptyError: Is empty
Traceback (most recent call last):
  File "/usr/share/comitup/comitup/nmmon.py", line 144, in reset_mdns
    mdns.add_hosts(dns_names)
  File "/usr/share/comitup/comitup/mdns.py", line 223, in add_hosts
    group.Commit()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.Avahi.IsEmptyError: Is empty`

I have tried purge and reinstall, no changes..

Thanks in advanced!

davesteele, 2021-09-29:
I suspect 9daacc0d2 will resolve the crash.

Can you use the connection from nmcli?

    sudo nmcli con
    sudo nmcli con up <name>

Are there "Exception encountered adding avahi record" messages in /var/log/comitup.log? If so, turn on debug logging to see more.

Ataraxiall, 2021-10-05:
Hello there,

sorry for the delay, but it seems that after some reboot, it works fine, it only gives me this error / warning if I look at the log.

warning: interface wlxa09f10bbe947 does not currently exist

But it works...