davesteele/comitup #194: no hotspot visible

fabmade, 2021-11-22:
Hi together,

i have a problem with the current comitup version.
i used the current comitup ISO. there i changed the hostname from "comitup" to an other one.
Also i modified the comitup.conf, i add here the <hostname> placeholder https://github.com/davesteele/comitup/blob/main/conf/comitup.conf

```
ap_name: <hostname>-<nn>
```
also i added to /etc/hostnames my new hostname (only the name without the number).

But after reboot i don't see the hotspot anymore. But if i run 'comitup -i" it shows me the correct name.

In the logs i found some errors:

**File "/usr/share/comitup/comitup/nm.p", line 148, in get_active_ip
addr = device.Ip4Config.Addresses[0][0]
IndexError: list index out of range**

Do you have an idea what could be the problem?

thanks
Fabian

davesteele, 2021-11-28:
I'm unable to replicate. Are there any special or non-ASCII letters in the hostname?

Are you on an IPv6-only network?

The fix is very likely d089e2

fabmade, 2021-12-13:
thank you davesteele, but no that was not the reason. in the meantime i was been able to find the cause of this problem.
The problem will happen if you install hamachi at the raspberry https://www.vpn.net/linux 
After that, comitup will not work anymore, the hotspot is not longer visible although comitup is in the state "HOTSPOT".

At the older version both programs worked together without that problems. but i don't know if a change in a new version of comitup is the problem or the new version of rasperry os. (hamachi version has not changed)

davesteele, 2021-12-13:
Comitup calls in NetworkManager, which takes over many services in place. They need to be disabled. I've not dealt with hamachi, but it could be dealing with one or more of those services.

Listed in the [wiki](https://github.com/davesteele/comitup/wiki/Installing-Comitup).

There should be entries in the comitup or syslog logs pointing to the error.

fabmade, 2021-12-14:
the services are still masked, and the config files are as you described. 

but yes sometimes i have an error if i start `comitup-cli` 
**ERROR:dbus.proxies:Introspect error on :1.19:/com/github/davesteele/comitup: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.**

Sometimes an error if i press (i)nfo:
**comitup.nm.py, line 96 in <listcomp>
.....
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnkonwnMethod: Keine derartige Schnittstelle "org.freedesktop.DBus.Properties" des Objekts im Pfad /org/freedesktop/NetworkManager/Devices/3**

But after reboot, only this error is in syslog, and comitup-cli don't show any errors again.

**File "/usr/share/comitup/comitup/nm.p", line 148, in get_active_ip
addr = device.Ip4Config.Addresses[0][0]
IndexError: list index out of range**

in the comitup log file are no errors.

davesteele, 2021-12-14:
That issue has been [addressed](https://github.com/davesteele/comitup/commit/d089e2aaaa0039db4eeb6a09f4c0e145d38e026c) in the repository.

closed 2022-05-13 by davesteele