davesteele/comitup #196: Issues with installation of package on Raspberry Pi OS Bullseye?

Alaric84, 2022-01-08:
Hey - I really like this project!
Are there any known issues installing on the bullseye based PiOs?
Thanks for your work!
Cheers 
Andreas

davesteele, 2022-01-08:
The [Comitup Image](https://davesteele.github.io/comitup/#Comitup%20Image) is based on Bullseye Raspberry Pi OS. It's use is recommended.

There are [installation instructions](https://github.com/davesteele/comitup/wiki/Installing-Comitup) for the deb install.

Alaric84, 2022-01-08:
Thanks for the quick answer - so the image is already up to date! Great! I think the base-image version would be worth mentioning. 
I already read the installation guide and really tried hard to follow each single step. I'll give it another try - or just switch to your image.

closed 2022-01-08 by Alaric84