davesteele/comitup #197: Include apt repository all or arm64 architecture support

gman77, 2022-01-12:
Hi,

The older apt repository comitup artifact (w/Ubuntu 20.04) was set to use "all" architecture.(1.10)

However I would like to use the latest release as recommended in the 
https://github.com/davesteele/comitup/wiki/Installing-Comitup

Would it be possible to specify "all" for the latest deb releases in the repo? or to include an arm64 artifact?

davesteele, 2022-01-13:
I'm confused - which package/version are you saying is not "all".

gman77, 2022-01-14:
Hi, 
the latest 1.31.deb when trying to install on ubuntu 64bit (raspberry pi) says arch not supported.

davesteele, 2022-01-16:
I've not tried Ubuntu, but all of the packages in the Comitup repository are "all".

gman77, 2022-01-16:
My bad I had missed a dependency.

closed 2022-01-16 by gman77