davesteele/comitup #199: Hotspot few minutes not visible after entering wrong wlan key

fabmade, 2022-02-02:
Hello davesteele,

I sadly have a problem with the hotspot.
After entering a wrong wifi password it sometimes takes 5 minutes or more until the hotspot is visible again.
I tried different versions 1.31, 1.32 and 1.32.
But this not happening every time, only sometimes.

In the logs i found this in daemon.log:

```
Feb  2 17:05:07 raspberrypi comitup[2401]: ERROR:dbus.service:Unable to append ([{'ssid': 'DSEIZ-DEV-TEST-FLE-402', 'strength': '100', 'security': 'unencrypted'}, {'ssid': '__FRITZ_BOX__', 'strength': '92.0', 'security': 'encrypted'}, {'ssid': '__FRITZ_BOX_GAST__', 'strength': '92.0', 'security': 'encrypted'}, {'ssid': 'HPNE', 'strength': '64.0', 'security': 'encrypted'}, {'ssid': 'FRITZ!Box 7490', 'strength': '48.0', 'security': 'encrypted'}, {'ssid': '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', 'strength': '48.0', 'security': 'encrypted'}, {'ssid': 'Internet', 'strength': '40.0', 'security': 'encrypted'}, {'ssid': 'Vodafone-EE31', 'strength': '40.0', 'security': 'encrypted'}, {'ssid': 'WLAN-571229', 'strength': '36.0', 'security': 'encrypted'}, {'ssid': 'Vodafone Hotspot', 'strength': '34.0', 'security': 'unencrypted'}],) to message with signature aa{ss}: <class 'ValueError'>: embedded null byte
Feb  2 17:05:07 raspberrypi comitup[2401]: ERROR:dbus.service:Unable to append ([{'ssid': 'DSEIZ-DEV-TEST-FLE-402', 'strength': '100', 'security': 'unencrypted'}, {'ssid': '__FRITZ_BOX__', 'strength': '92.0', 'security': 'encrypted'}, {'ssid': '__FRITZ_BOX_GAST__', 'strength': '92.0', 'security': 'encrypted'}, {'ssid': 'HPNE', 'strength': '64.0', 'security': 'encrypted'}, {'ssid': 'FRITZ!Box 7490', 'strength': '48.0', 'security': 'encrypted'}, {'ssid': '\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', 'strength': '48.0', 'security': 'encrypted'}, {'ssid': 'Internet', 'strength': '40.0', 'security': 'encrypted'}, {'ssid': 'Vodafone-EE31', 'strength': '40.0', 'security': 'encrypted'}, {'ssid': 'WLAN-571229', 'strength': '36.0', 'security': 'encrypted'}, {'ssid': 'Vodafone Hotspot', 'strength': '34.0', 'security': 'unencrypted'}],) to message with signature aa{ss}: <class 'ValueError'>: embedded null byte
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/_compat.py", line 39, in reraise
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     raise value
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1950, in full_dispatch_request
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     rv = self.dispatch_request()
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1936, in dispatch_request
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     return self.view_functions[rule.endpoint](**req.view_args)
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/share/comitup/web/comitupweb.py", line 84, in index
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     points = cached_points()
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/cachetools/decorators.py", line 22, in wrapper
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     v = func(*args, **kwargs)
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/share/comitup/web/comitupweb.py", line 71, in cached_points
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     return ciu_client.ciu_points()
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     return self._connection.call_blocking(self._named_service,
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     reply_message = self.send_message_with_reply_and_block(
Feb  2 17:05:07 raspberrypi comitup-web[3639]: dbus.exceptions.DBusException: org.freedesktop.DBus.Python.ValueError: Traceback (most recent call last):
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/dbus/service.py", line 755, in _message_cb
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     _method_reply_return(connection, message, method_name, signature, *retval)
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/dbus/service.py", line 256, in _method_reply_return
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     reply.append(signature=signature, *retval)
Feb  2 17:05:07 raspberrypi comitup-web[3639]: ValueError: embedded null byte
Feb  2 17:05:07 raspberrypi comitup-web[3639]: [2022-02-02 17:05:06,938] ERROR in app: Exception on / [GET]
Feb  2 17:05:07 raspberrypi comitup-web[3639]: Traceback (most recent call last):
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 2447, in wsgi_app
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     response = self.full_dispatch_request()
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1952, in full_dispatch_request
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     rv = self.handle_user_exception(e)
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1821, in handle_user_exception
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     reraise(exc_type, exc_value, tb)
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/_compat.py", line 39, in reraise
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     raise value
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1950, in full_dispatch_request
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     rv = self.dispatch_request()
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1936, in dispatch_request
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     return self.view_functions[rule.endpoint](**req.view_args)
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/share/comitup/web/comitupweb.py", line 84, in index
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     points = cached_points()
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/cachetools/decorators.py", line 22, in wrapper
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     v = func(*args, **kwargs)
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/share/comitup/web/comitupweb.py", line 71, in cached_points
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     return ciu_client.ciu_points()
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     return self._connection.call_blocking(self._named_service,
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     reply_message = self.send_message_with_reply_and_block(
Feb  2 17:05:07 raspberrypi comitup-web[3639]: dbus.exceptions.DBusException: org.freedesktop.DBus.Python.ValueError: Traceback (most recent call last):
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/dbus/service.py", line 755, in _message_cb
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     _method_reply_return(connection, message, method_name, signature, *retval)
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/dbus/service.py", line 256, in _method_reply_return
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     reply.append(signature=signature, *retval)
Feb  2 17:05:07 raspberrypi comitup-web[3639]: ValueError: embedded null byte
Feb  2 17:05:07 raspberrypi comitup-web[3639]: [2022-02-02 17:05:07,017] ERROR in app: Exception on / [GET]
Feb  2 17:05:07 raspberrypi comitup-web[3639]: Traceback (most recent call last):
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 2447, in wsgi_app
Feb  2 17:05:07 raspberrypi comitup-web[3639]:     response = self.full_dispatch_request()
Feb  2 17:05:07 raspberrypi comitup-web[3639]:   File "/usr/lib/python3/dist-packages/flask/app.py", line 1952, in full_dispatch_request
[... cut ...]
2022-02-02 17:03:59,828 - comitup - INFO - Setting state to HOTSPOT
2022-02-02 17:03:59,903 - comitup - INFO - Activating hotspot
2022-02-02 17:04:02,946 - comitup - ERROR - Exception encountered adding avahi record
2022-02-02 17:04:03,623 - comitup - ERROR - Exception encountered adding avahi record
```

Also i have sometimes problems if the wlan connection is not aviable for few minutes, after that, the reconnection won't work

Perhaps you have an idea? I use raspberry pi zero and your last ISO, no other modifications.

Thanks
Fabian

davesteele, 2022-02-03:
It looks like you are seeing #198. The fix is in the repo, but not yet released.

fabmade, 2022-02-03:
ok thank you very much, it looks better now!

closed 2022-02-03 by fabmade

davesteele, 2022-02-03:
[v1.34-1](https://davesteele.github.io/comitup/archive.html) is released, and is part of the latest [image](https://davesteele.github.io/comitup/). Fixes #198.