davesteele/comitup #200: docker

sebastianeferrer, 2022-02-09:


davesteele, 2022-02-09:
If the issue involves networking, I suspect the solution follows along the lines of [this](https://davesteele.github.io/raspberrypi/2021/04/29/comitup-and-pihole/).

sebastianeferrer, 2022-02-15:
Sorry Dave, I'd posted this issue by mistake, but now I'm editing first comment to explain what it came about. It's about conflict between docker and comitup.

davesteele, 2022-02-16:
Good work.

The add_hosts() function is getting a list of network devices for publishing ZeroConf services and mdns records - it needs to list both wired and wireless devices. I can believe that it could have some heartache with devices created by Docker.

I suspect a new function (get_network_endpoints()?) that filters a la [get_wifi_devices()](https://github.com/davesteele/comitup/blob/1bf4a03e8302058fd57c11210335e1b97a86579c/comitup/nm.py#L84) to provide a list of "[useful](https://developer-old.gnome.org/NetworkManager/stable/nm-dbus-types.html#NMDeviceType)" devices.

davesteele, 2022-02-16:
Note that Comitup also takes action on loss of a wifi connection.