davesteele/comitup #201: Apache2 and Comitup

alf4all, 2022-02-20:
On a Raspberry Pi 3B with OS 32-bit I have installed apache2 before installing comitup. Now I can access the apache2 default page via WiFi but how can I make a setup to an existing AP? I it by a webpage or by SSH? did I miss something?

alf4all, 2022-02-20:
So, reading "Bootstrap Wifi using Wifi" ther must be a comitup-web somewhere...

davesteele, 2022-02-20:
Comitup and Apache both want/need ownership of port 80 - they can't both run at the same time. See [this example](https://davesteele.github.io/raspberrypi/2021/04/29/comitup-and-pihole/) on how to get such services to coexist. This gives ownership of the port to Comitup when the upstream Wifi connection can't be made.

There are some other potential [complications](https://github.com/davesteele/comitup/wiki/Installing-Comitup) involved with installing Comitup as a package. You may want to start with the [Comitup image](https://davesteele.github.io/comitup/). It is just Raspberry Pi OS with Comitup installed.

alf4all, 2022-02-21:
Hai Dave, I am now working on the Pi-hole-Comitup example adapted for Apache2. I have not seen the GUI of Comitup yet, but is there an option to choose for a non-connected start of the application? Or do I have to solve that situation in comitup-callback? Thanks for this fantastic application and good service. Regards...

alf4all, 2022-02-21:
Small remark: Advanced options of Raspberry Pi Imager (Set hostname) do not override the default comitup-NNN of Comitup.

davesteele, 2022-02-21:
Start with Apache2 disabled, then you should see the GUI (if upstream is not connected at startup). Then configure the callback to start Apache2 when needed.

Comitup will determine the state to start in automatically. If upstream is already connected, the GUI will never come up (single adapter mode).

closed 2024-07-10 by davesteele