davesteele/comitup #203: Comitup 1.35 dbus errors. Unable to get to web UI

Geektoolkit, 2022-05-03:
I'm on a pi4, just resinstalled things. I had a version running a while back, revisited some code I had that used it and started hitting issues. So I apt removed it, and then followed the instructions to get the latest from the depot, ended up with a 1.35 build.

The AP shows up right off and my iphone xs can connect to it no problem, and seems to stay connected. But the web page doesn't come up, and if i go to http://10.41.0.1 which usually works, I get 'could not connect to the server' and no webpage shows up.  On the pi if I do sudo comitup-cli I get a dbus error that it can't connect.

The logs show a bunch of process exits and restarts which look like it is crashing.  The traceback shows Comitup-web, line 33.  Ther eis a dbus.proxies: Introspect error  (did not recieve a reply).

Running comitup at a console will occasionally say 'ok' but for the most part I don't get any output.  

sudo comitup-web also gives the dbus error (did not recieve a reply)

I'm new to Linux so I just am not sure where to go to get you the info you'd need to debug this. Please let me know what i can provide to help debug this.

davesteele, 2022-05-03:
There is a recent [bug](https://github.com/seveas/python-networkmanager/issues/93) in the python3-networkmanager package, which causes Comitup to crash on startup (Look for comitup "Traceback"s in /var/log/syslog). There is a [patched version](https://davesteele.github.io/comitup/deb/python3-networkmanager_2.2-1dss01_all.deb) of the package in the archive, but it is not yet reflected in the repository (or significantly tested).

Geektoolkit, 2022-05-04:
When I try to extract that file it seems like it gets corrupted or something when I go to install it. I get a 'exclusive lock on the packaging backend' even after a reboot.  When I try dkpgr it says the archive uses an unknown compression for member 'control.tar.zst', giving up.

Is this because I'm a linux newb? I'll research those but hoping that it's just a corrupt package

davesteele, 2022-05-04:
I built it on Ubuntu, on the road. Apparently, Ubuntu debs are currently incompatible with Debian. There's a newer version in the repo and in the [archive](https://davesteele.github.io/comitup/latest/python3-networkmanager_latest.html).

Glad to hear you got it working.

Geektoolkit, 2022-05-09:
Dave I was going to close this one but I don't know if the python fix has made it into the main branch. Please close when this isn't an issue. I figure leaving it open may help others that hit the issue for now.  I hit a similar issue that I believe is new and opened up a new bug for it (and provided better logs and info)

closed 2022-05-10 by davesteele