davesteele/comitup #207: Can't change SSID of AP...

ass805, 2022-05-12:
Hello,

My problem is, that I changing app_name in /etc/comitup.conf file, but AP name still is comitup-xx. I tried to reinstall Comitup, but nothing works, maybe this functions is not available any more?

davesteele, 2022-05-12:
Be sure that the variable you are changing in /etc/comitup.conf is "ap_name" - that's "ap" an in "access point".

ass805, 2022-05-12:
[image]

That's how looking my config file. I was trying to change "ap_name: Nothing" and it still showing as "comitup-24" after reboot.
Maybe there is more I can show you to detect this problem?
Also, I have a question about HTML page editting, is it possible when comitup already installed in Pi? Where I can find Web interface files?

I'm using: Raspberry Pi Zero W + Debian Buster Raspbian OS Desktop

GrandChriss, 2022-05-12:
I solved that problem. Changed ap_name in /usr/share/comitup/comitup/comitup.py file. 

I have another question, where I can find command when after reboot checking Wi-Fi and if it's connected it kill's comitup hotspot and web interface? I need to add some command there. Is it possible?

Thank you for this project! It really helps me

davesteele, 2022-05-12:
If you have a single WiFi interface, the hotspot and web service will automatically be killed once connected upstream.

If there is something else you need to do, see _external_callback_ in "man 5 comitup-conf".

davesteele, 2022-05-12:
Just saw your screen shot. The variable definition is commented out (to show the default). Remove the leading "# ".

GrandChriss, 2022-05-12:
I mean, maybe there's possibility to check wifi connection and start command if connected?

davesteele, 2022-05-12:
Yes, look up _external_callback_.

davesteele, 2022-05-12:
Something like:

    # cat /usr/local/bin/comitup-callback
    #!/bin/bash
    
    if  [[ "$1" == "CONNECTED" ]] ; then
        ...
    fi
    
    # chmod 755 /usr/local/bin/comitup-callback

GrandChriss, 2022-05-13:
So what I should do? How launch bash script after callback CONNECTED?

davesteele, 2022-05-13:
I don't have any other guidance here.

closed 2022-05-13 by GrandChriss